Hypfer/Valetudo #305: Feature request Static IP

Wessix, 2019-08-12:
Hi, would it be possible to add a possibility to assign static ip to the robot via the webinterface.
In my setup this would ease the use cause i can't assign it to the MAC with the Telekom Router i have to use.

If not is there a workaround via the ssh console i can use to give the robot a permanent static ip i want it to have?

r1sim, 2019-08-15:
Have you tried using the hostname of the robot instead of the ip? If your robot has the default hostname, you should be able to access the robot using http://rockrobo/.

If you really need a static ip you would have to do this using ssh. The robot is running ubuntu, so theoretically you can assign a static ip by editing the /etc/network/interfaces file.
I don't recommend doing this if you don't know what you are doing. Wifi is the only way to talk to the robot (without opening it) and you may risk locking yourself out with an invalid network configuration. I'm also not sure if this file will be replaced when you do a factory reset.

samzal, 2019-09-06:
>  you may risk locking yourself out with an invalid network configuration
This.
Editing /etc/network/interfaces doesn't seem to be that easy anyway, as it is empty on my robo. Also I have no systemd installed apparently. If you are not 100% sure what you are doing dont risk it. You could always buy a better router and assign IPs there.

marv42dp, 2019-11-10:
I second this request, because it would be a workaround for the DHCP bug I've discovered (robo won't get an IP if the DHCP server doesn't run on the WIFI AP).

The reset WIFI function should not be affected by this, so you can always reset the WIFI and start over, if you managed to lock yourself out.

mxwx-droid, 2019-12-11:
As far as I can say you can configure valetudo to have a static ip (additionally) but you have to do it by ssh. Well, here we go...

Valetudo is currently getting its ip address by dhclient - as usual when using ubuntu. In general it should be possible to edit /etc/network/interfaces, as proposed before, but that could disturb the mechanics of dhclient - my S5 will not start network anymore if manipulation /etc/network/interfaces. More compatible to dhclient is to use the dhclient.config - usually found at /etc/dhcp/dhclient.conf

In dhclient.conf there is a commented part which is starting with:

#alias {
#  interface "eth0";
#  fixed-address 192.5.5.213;
#  option subnet-mask 255.255.255.255;
#}

uncomment this part and change it to your needs (e.a. change interface "eth0" to "wlan0", fixed-address and option subnet-mask corresponding to your other network configuration,) and after next reboot you (normaly) have two ip addresses bound to your wlan0 interface - a dynamic one that is assigned by dhcp and additionally a static one that is configured just here. The dynamic one is not assigned if there is no dhcp-server in reach.

Hypfer, 2020-04-12:
This is out-of-scope for the project

closed 2020-04-12 by Hypfer