Hypfer/Valetudo #308: Setting maximum charging level

Ph1r3ph0x, 2019-08-21:
Hi everyone :)
I think a lot of people don't actually use the whole battery capacity the roborock provides, even when doing the whole flat in maximum mode. It would therefore be really nice to have the ability to set a maximum charging level in order to gain some battery life! Li-Ion batteries live way longer if you don't use the lower and upper 10% charging levels.
There is already a thread in the dustcloud repo: [https://github.com/dgiese/dustcloud/issues/188](url). 
I found a "StopCharging" function in the RoboController. If somebody here knows a way to call this function then that might be a way to add this feature :)

acidg, 2019-09-17:
Very cool idea! Would also love this feature :)

adfx, 2020-01-22:
one up here, too! Nice idea and very welcome to have this feature available.

Hypfer, 2020-04-12:
This is probably out of scope and currently impossible.

If this changes, feel free to reopen the issue

closed 2020-04-12 by Hypfer