Hypfer/Valetudo #310: No mqtt and map

jlo88, 2019-08-22:
I followed the guide to flash my gen1 vacuum cleaner. I started by doing a factory reset and installing firmware v11_003468.fullos.pkg. I build the firmware by: 

```
rockrobo/dustcloud/devices/xiaomi.vacuum/firmwarebuilder/imagebuilder.sh \
     --firmware=rockrobo/firmware/v11_003468.fullos.pkg \
     --soundfile=rockrobo/firmware/english.pkg \
     --public-key=/root/.ssh/id_ed25519.pub \
     --valetudo-path=rockrobo/valetudo \
     --disable-firmware-updates \
     --replace-adbd 
```

and flashed it by: 
`mirobo --ip 192.168.8.1 --token secret_token update-firmware output/v11_003468.fullos.pkg
`
This works well and I can access the vacuum via the browser over 192.168.8.1 as expected, I can see the map and I set the wifi to connect it to my own network. The vacuum has IP address 192.168.0.31

Next I edit the Valutedo configuration file to setup mqtt for home assistant according to these instructions: https://github.com/Hypfer/Valetudo/wiki/Home-Assistant-Integration I edited `/mnt/data/valetudo/config.json` to: 
```
{
  "spots": [],
  "areas": [],
  "mqtt": {
    "enabled": false,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "mqtt://my_username:my_password@my_broker_IP",
    "provideMapData": true,
    "caPath": ""
  },
  "dummycloud": {
    "spoofedIP": "203.0.113.1",
    "bindIP": "127.0.0.1"
  },
  "httpAuth": {
    "enabled": false,
    "username": "valetudo",
    "password": "valetudo"
  },
  "allowSSHKeyUpload": true,
  "map_upload_host": "http://127.0.0.1"
}

```

I use mqtt explorer to check the messages but nothing seems to come by. I want to integrate the map in hassio by using the addon ICantBelieveItsNotValetudo. I also cannot access the map via `http://192.168.0.31/api/remote/map`, I get `Cannot GET /api/remote/map`. 

Furthermore I have the vacuum control integrated in home assistant via 

```
vacuum:
  - platform: xiaomi_miio
    host: !secret xiaomi_ip
    token: !secret xiaomi_token
```
This integration works but I cannot seem to get the map working nor the MQTT messages. `iptables -L` shows: 

```
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
DROP       udp  --  anywhere             anywhere             udp dpt:6665
DROP       tcp  --  anywhere             anywhere             tcp dpt:6665

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
REJECT     all  --  anywhere             ot.io.mi.com         reject-with icmp-port-unreachable
```

Am I missing something here? 
Valetudo version: 0.4.0
Firmware version: 3.3.9
Firmware build: 003468

Hypfer, 2019-08-24:
```
  "mqtt": {
    "enabled": false,
```

jlo88, 2019-08-24:
Okay, that was not very smart of me, thanks! 
I changed this and restarted valetudo via `service valetudo restart` but I still don't get anything on mqtt. 

I don't have to specify the port in the broker URL right? I tried both with and without port. Is there maybe a log that I can check somewhere?

Hypfer, 2019-08-24:
https://github.com/Hypfer/Valetudo/wiki/FAQ#where-do-i-find-the-log-file

This is the expected logline
https://github.com/Hypfer/Valetudo/blob/master/lib/MqttClient.js#L154

jlo88, 2019-08-26:
Thank you for the links, I looked at the logfile and this is what comes by when I restart valetudo: 
```
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
```

The configuration file: 
```
{
  "spots": [],
  "areas": [],
  "mqtt": {
    "enabled": true,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "mqtt://my_username:my_pw@192.168.0.16",
    "provideMapData": true,
    "caPath": ""
  },
  "dummycloud": {
    "spoofedIP": "203.0.113.1",
    "bindIP": "127.0.0.1"
  },
  "httpAuth": {
    "enabled": false,
    "username": "valetudo",
    "password": "valetudo"
  },
  "allowSSHKeyUpload": true,
  "map_upload_host": "http://127.0.0.1"
}

```

Hypfer, 2019-08-26:
Check the nat table

`iptables -t nat -L`

Should look like this:
```
Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination
DNAT       tcp  --  anywhere             ot.io.mi.com         tcp dpt:http to:127.0.0.1:8053
DNAT       udp  --  anywhere             ot.io.mi.com         udp dpt:8053 to:127.0.0.1:8053
```

Also make sure that your `/etc/hosts` contains the relevant hostnames.

jlo88, 2019-08-26:
`iptables -t nat -L` returns:
```
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
DNAT       tcp  --  anywhere             ot.io.mi.com         tcp dpt:http to:127.0.0.1:8053
DNAT       udp  --  anywhere             ot.io.mi.com         udp dpt:8053 to:127.0.0.1:8053

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination  
```
`/etc/hosts` contains: 
```
127.0.0.1       localhost
127.0.1.1       rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

0.0.0.0       awsbj0-files.fds.api.xiaomi.com
0.0.0.0       awsbj0.fds.api.xiaomi.com
### VALETUDO HOSTS INIT ###
###########################################################
### VALETUDO HOSTS
###########################################################
#
# Add to /etc/hosts
#

203.0.113.1  ot.io.mi.com
2001:db8::1  ot.io.mi.com
203.0.113.1  ott.io.mi.com
2001:db8::1  ott.io.mi.com
203.0.113.1  us.ot.io.mi.com
2001:db8::1  us.ot.io.mi.com
203.0.113.1  us.ott.io.mi.com
2001:db8::1  us.ott.io.mi.com
203.0.113.1  sg.ot.io.mi.com
2001:db8::1  sg.ot.io.mi.com
203.0.113.1  sg.ott.io.mi.com
2001:db8::1  sg.ott.io.mi.com
203.0.113.1  st.ot.io.mi.com
2001:db8::1  st.ot.io.mi.com
203.0.113.1  st.ott.io.mi.com
2001:db8::1  st.ott.io.mi.com
203.0.113.1  de.ot.io.mi.com
2001:db8::1  de.ot.io.mi.com
203.0.113.1  de.ott.io.mi.com
2001:db8::1  de.ott.io.mi.com
203.0.113.1  ru.ot.io.mi.com
2001:db8::1  ru.ot.io.mi.com
203.0.113.1  ru.ott.io.mi.com
2001:db8::1  ru.ott.io.mi.com
203.0.113.1  in.ot.io.mi.com
2001:db8::1  in.ot.io.mi.com
203.0.113.1  in.ott.io.mi.com
2001:db8::1  in.ott.io.mi.com
203.0.113.1  tw.ot.io.mi.com
2001:db8::1  tw.ot.io.mi.com
203.0.113.1  tw.ott.io.mi.com
2001:db8::1  tw.ott.io.mi.com
203.0.113.1  ea.ot.io.mi.com
2001:db8::1  ea.ot.io.mi.com
```

jlo88, 2019-09-07:
@Hypfer, sorry for asking but do you maybe have some ideas on what to do? I'd really like to get the map functionality working!

bartekd123, 2019-09-14:
I am getting the same thing, its not discovering in homeassistant. 
IN the log file, I see:
```
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Connected successfully to mqtt server
Robot connected

```

So it looks like it is connecting to MQTT, but not much after that. My MQTT server is not the same one built into home-assistant, it is outside of it. Does that matter? @jlo88  is your mqtt server the one in HomeAssistant?

bartekd123, 2019-09-15:
Nevermind, it showed up, even though when I monitor using mqtt explorer it doesn't show any data. I looked in my list of devices, and shows vacuum.rockrobo. Still can't get map to show, but at least I am able to control from home assistant.

jlo88, 2019-09-15:
@bartekd123 I'm using mosquito via hassio. 
I'm also able to control the vacuum cleaner via home assistant but directly via the token. I think that's also how it works with you at the moment.

jlo88, 2019-09-15:
I hosted another mqtt broker via mosquito on a laptop and then the vacuum is able to connect to it..

jlo88, 2019-09-15:
@bartekd123 @Hypfer  Okay, I figured out the problem, my password ends on a question-mark. I've reproduced the problem with second mqtt broker and when I remove the question-mark at the end of password it works. Any ideas on how to get around this or should I remove the question-mark and reconfigure my other mqtt devices?

closed 2019-10-26 by Hypfer