Hypfer/Valetudo #312: Maps not loading on iOS browsers

talondnb, 2019-08-25:
As per https://github.com/Hypfer/Valetudo/issues/295, I'd like to open this issue. I'm unable to load the Map section under iOS in both Safari and Chrome browsers. I've also tried clearing cache and enabling all experimental features in Safari settings but the map still won't load.

Running iOS 12.4 and latest Valetudo on Gen2 S55.

Admonstrator, 2019-09-07:
I can confirm this issue.

Running iOS 12.4.1 + Valetudo 0.4.0 on FW 3.3.9 (Build 001720)

BirokratskaZila, 2019-09-07:
Same issue for me.

iOS 12.4 + Valentudo 0.4.0 on FW 3.3.9 (Build 003468)

nikolaykasyanov, 2019-10-11:
I'm observing the same behaviour on Safari for macOS as well. Console gives a clue:

```
[Error] Blocked http://rockrobo/zone/js-modules/vacuum-map.js from asking for credentials because it is a cross-origin request.
[Error] Failed to load resource: the server responded with a status of 401 (Unauthorized) (vacuum-map.js, line 0)
[Error] Failed to load resource: the server responded with a status of 401 (Unauthorized) (icon.png, line 0)
```

onkelandy, 2019-10-17:
I can see the map with latest OS and Valetudo. Have to drag my finger over the whole screen to zoom out. Problem: not possible to pan/move the map and zoom center is wrong

onkelandy, 2019-10-18:
Just tried it with iOS 13 and it seems to work on an iPhone 7 in Safari

ottelo9, 2019-11-05:
I've problems that the map doesnt refresh automatically in chrome for PC. And I've the same problems with iOS not showing the map on iOS12.

ottelo9, 2019-11-13:
Update:
After disable HTTP Login the map is displayed on my iOS devices! So its a problem with http login.

talondnb, 2019-11-13:
I've also tested this, iOS 13.2.2 and I've just disabled http login and it's now showing the map. Thanks @ottelo9

LasseSLambertsen, 2019-11-28:
Same issue here, disabling http auth fixes it.
Though not a solution for me as it is accessible over internet

Hypfer, 2020-05-15:
This won't be fixed unless someone with an Apple devices opens a PR.

IzK666, 2020-05-20:
Today is my first day with valetudo.
I'm having the same issue, but nothing to do with IOS: Just any "lite" browser in android.
I use Lightning browser, but I tested on Opera Lite and Firefox Lite with the same result: The map isn't loading at all. I tried to create a new zone but the map is blank and can't draw anything.

Hypfer, 2020-05-20:
I just tried the Opera Mini Beta on android and got a map rendered.
It was rather slow though.
Same goes for Lightning Browser. That one is quicker though.
> Speed - By utilizing the WebKit rendering engine that comes built into your Android device, Lightning can ensure a swift, lightweight experience.

I'm using Android 9 so the WebView might be newer than yours.

Can't test Firefox lite since it's not avaliable in Germany, sadly :/

Anyways. These are probably unrelated issues.

IzK666, 2020-05-20:
You're right. The issue is my android 7 (WebView 60). After update I can see the map.
Thanks.

Hypfer, 2020-08-06:
252c22d might've fixed this

closed 2021-01-12 by Hypfer