Hypfer/Valetudo #328: Custom Commant via MQTT dosn't work

kreativmonkey, 2019-09-23:
I tried to start Zoned Cleanup within a automation in Home Assistant. When the Automation is called, Home Assistant sends an MQTT to the valetudo/rockrobo/custom_command topic with the following payload:

```
{
  "command" : "zoned_cleanup",
  "zone_ids" : [ "Kitchen" ]
}
```

the Area "Kitchen" is inside the config.json file. I can start my Xiaomi Roborock S5 (with Valetudo 0.4) within Home Assistant but can not start any Zone Cleanup. If i use the Valetudo Webfrontend everything works fine with the Kitchen zone cleaning.

I enabled the Home Assistant integration and that is what my Roborock is publishing to Home Assistant:

```
{
  "name" : "rockrobo",
  "supported_features" : [ "turn_on", "pause", "stop", "return_home", "battery", "status", "locate", "clean_spot", "fan_speed", "send_command" ],
  "command_topic" : "valetudo/rockrobo/command",
  "battery_level_topic" : "valetudo/rockrobo/state",
  "battery_level_template" : "{{ value_json.battery_level }}",
  "charging_topic" : "valetudo/rockrobo/state",
  "charging_template" : "{{value_json.charging}}",
  "cleaning_topic" : "valetudo/rockrobo/state",
  "cleaning_template" : "{{value_json.cleaning}}",
  "docked_topic" : "valetudo/rockrobo/state",
  "docked_template" : "{{value_json.docked}}",
  "error_topic" : "valetudo/rockrobo/state",
  "error_template" : "{{value_json.error}}",
  "fan_speed_topic" : "valetudo/rockrobo/state",
  "fan_speed_template" : "{{ value_json.fan_speed }}",
  "set_fan_speed_topic" : "valetudo/rockrobo/set_fan_speed",
  "fan_speed_list" : [ "min", "medium", "high", "max" ],
  "send_command_topic" : "valetudo/rockrobo/custom_command",
  "json_attributes_topic" : "valetudo/rockrobo/attributes"
}
```
the same is happend with go_to... 

I used the [MqttClient.js file](https://github.com/Hypfer/Valetudo/blob/22be426ac0f24b61ecbb8a099e5ba075781cfae3/lib/MqttClient.js ) to find out what to do but i can't find any working option! 

**Vacuum Model:** S5

kreativmonkey, 2019-09-23:
Sorry i forgot, nothing to find in the valetudo.log file!

hkemal, 2019-09-24:
Hi,

I think I have the same issue using Node-Red and Domoticz with Mosquitto.
Valetudo version is 0.3.1.

I am sending commands to valetudo/rockrobo/command topic.
The robot is only responding to "turn_on", "stop" and "locate".
"pause" and "return_home" do not get any response.
Did not try the zone cleaning until now, that's the next step.

hkemal, 2019-09-27:
So I gave the zoned cleaning a try today.
This is what I am sending with Node-Red. But the robot does nothing.

valetudo/rockrobo/custom_command : msg : Object
object
topic: "valetudo/rockrobo/custom_command"
payload: object
command: "zoned_cleanup"
zone_ids: array[1]
0: "Kitchen"
qos: 0
retain: false
_topic: "domoticz/out"
_msgid: "18ddbdaf.bb27b2"

hkemal, 2019-09-28:
I updated the valetudo to version 0.4.0 and now everything is working fine :)!

fillilutten, 2019-10-03:
> So I gave the zoned cleaning a try today.
> This is what I am sending with Node-Red. But the robot does nothing.
>
> valetudo/rockrobo/custom_command : msg : Object
> object
> topic: [...]

I want to try and do the same. Could you upload the nodes to do this? Can't get it to work.

hkemal, 2019-10-03:
Hi,
this is my whole Node-Red flow for the interface between Valetudo and Domoticz.

[valetudo_domoticz.json.zip](https://github.com/Hypfer/Valetudo/files/3687151/valetudo_domoticz.json.zip)

fillilutten, 2019-10-03:
Oh sorry missed that you used Domoticz, i'm running Home Assistant. Huge thanks anyway!

kreativmonkey, 2019-10-05:
Sorry, i don't know why but the update to Valetudo 0.4 dosn't work correct, after reinstallation my mqtt works fine. @fillilutten i use Home Assistant too and you need to create a script and use this script in your automation, i don't know why but calling the vaccum.send_command from the automation dosn't work for me.

closed 2019-10-05 by kreativmonkey