Hypfer/Valetudo #331: Installation Instructions - location of patched firmware wrong?

markusloffler, 2019-09-28:
The installation instructions just have been changed with the comment "Fix directory structure to be consistent with the explanations above. The image is built in the `firmware` dir, not `image`"

They now state that the flashing command is:

mirobo --ip 192.168.8.1 --token XXXXXXXXXXXXXXXX update-firmware firmware/output/v11_001748.fullos.pkg

Is this really correct? For me, the firmware file is still created in image/output...

rassaei, 2019-09-29:
Hi, I just looked into it:
With
"Die Befehle 'mkdir image' und 'cd image' sind nicht mehr aktuell"
@Tschipel committed on 12 Aug
he removed the "mkdir image" and "cd image" from the instructions, without adjusting the line in the  mirobo update command.
That command got updated in the change that you mentioned, so the instructions are consistent again. As far as I can see the instructions were wrong for about one month.

However, the image is created in your current working directory. If you create a directory "image" and change into it before running the imagebuilder, it is created in that directory. But if you follow the instructions, you should be in the "firmware" directory when you run the imagebuilder script.

markusloffler, 2019-09-30:
You are right. I totally missed the removal of the mkdir image/cd image commands. Thanks!

closed 2019-09-30 by markusloffler