Hypfer/Valetudo #332: Vacuum woke up without timer and spoke in different language

stevenroose, 2019-09-30:
So I'm not home for a week but my girlfriend reported some quite disconcerting event yesterday. Our Valetudo is configured in English and she always speaks English. We also never use the timer functions because I usually just turn it on when I go for lunch (I work at home).

So apparently yesterday night (around 3:00 UTC) she woke up and spoke in a language my girlfriend could not recognize. She woke up from it and after several minutes, she spoke again. My gf speaks Portuguese, English, Spanish and French and would certainly recognize German, Dutch and any Scandinavian language. So I'm thinking it would have been Chinese.

Our home network is normal, behind a NAT without any ports open. We don't have any MQTT devices turned on, we just use the web interface using the local IP. I'd be surprised if anyone would have been able to enter our network. (I do run a small server with some Bitcoin stuff on it but that's all behind a Tor hidden service, so should really not be reachable.)

So my Q would be if anyone has ever heard of a similar experience. A week or two ago, I updated to the new 0.4.0 software.

ckruetze, 2019-09-30:
I haven't seen it in person but I have seen discussions where people mention that the vacuum factory reset itself.
That would lead to it talking Chinese with you. Have a look if Valetudo is gone and the machine is reset to factory defaults.

rand256, 2019-10-01:
Your unique issue isn't that unique. See #206.

stevenroose, 2019-11-04:
Closing as it's a duplicate of #206.

closed 2019-11-04 by stevenroose