Hypfer/Valetudo #337: Hassio no vacuum.valetudo over mqtt

dennis071nl, 2019-10-07:
I cannot operate the vacuum through HomeAssistant (Hassio 0.99.3)

**To Reproduce**

**Expected behavior**

**Screenshots**

**Vacuum Model:** 

**Valetudo Version: ** 

**User-Agent** 

**Additional context**

oechslein, 2019-10-17:
Do you have an entry in your logfile with
ERROR (MainThread) [homeassistant.components.mqtt.vacuum] Exception in async_discover when dispatching 'mqtt_discovery_new_vacuum_mqtt': ({'name': 'rockrobo', 'schema': 'state', 'supported_features': ['start', 'pause', 'stop', 'return_to_base', 'battery', 'status', 'locate', 'clean_spot', 'fan_speed', 'send_command'], 'command_topic': 'valetudo/rockrobo/command', 'state_topic': 'valetudo/rockrobo/state', 'set_fan_speed_topic': 'valetudo/rockrobo/set_fan_speed', 'fan_speed_list': ['whisper', 'min', 'medium', 'high', 'max', 'mop'], 'send_command_topic': 'valetudo/rockrobo/custom_command', 'json_attributes_topic': 'valetudo/rockrobo/attributes', 'platform': 'mqtt'},)
?

I supect there is currently a mismatch between the autoconfig message what valetudo sends ([after this change](https://github.com/Hypfer/Valetudo/commit/92d7285b2a3bf76096ce86fb5a7f823f133960b8)) and what home assistant expects (v0.100.2).
Difference is "return_home" versus "return_to_base" in the supported_features list.
Maybe @TheRealMoeder can comment on this?

dennis071nl, 2019-10-17:
where can i find the logfile??

oechslein, 2019-10-17:
Sorry I meant the home assistant logfile "home-assistant.log".

oechslein, 2019-10-18:
And try replacing in lib/MqttClient.js line 106 the "return_to_base" back to "return_home" an see if that helps like it helped myself.

dennis071nl, 2019-10-18:
on the vacuum, i log in over ssh, in the /lib directory there is no
MqttClient.js....??

oechslein, 2019-10-18:
> on the vacuum, i log in over ssh, in the /lib directory there is no MqttClient.js....?? 
> Or should i compile it after adapting the file...? 
Yes you would need to recompile it.
Have you looked into your home assistant log? If you have the same output like above, I can create a change to fix this issue.

dennis071nl, 2019-10-18:
got it! apparently my mistake that i activated both mqtt addon and embedded
broker. vacuum.roborock is alive! :-)

bonze4, 2019-10-21:
@dennis071nl Did you rename the MqttClient.js file? I have the same problem. My robot connects well to the broker but i can not see the vacuum in HA. The discover feature does not work for now.

dennis071nl, 2019-10-21:
Tim, which mqtt broker do you use?? I used both the addon as weas the em
edded broker.. Just use the addon, worked for me...

bonze4, 2019-10-22:
@dennis071nl I don’t use the embedded broker. I checked my configuration.yaml and there is no entry for the embedded one. I’ve installed the mosquito addon. The broker works fine and already discovered some Tasmota devices. So basically everything works fine. I don’t no why it’s not possible for HA to discover the vacuum.

dennis071nl, 2019-10-22:
Tim, with the addon enabled, there is no need for any reference to mqtt in
configuration.yaml...

That did the trick for me...

oechslein, 2019-10-22:
@bonze4 Did you compile valetudo yourself (from lastest master)?
Search for "ERROR (MainThread) [homeassistant.components.mqtt.vacuum]" in the homeassistant.log (in your home assistant config folder).

bonze4, 2019-10-23:
@oechslein I don't compiled Valetudo by myself. I'm still running 0.4.0 because i thought that @dennis071nl may have an idea how to fix the problem.

I have no log-entry related to "ERROR (MainThread) [homeassistant.components.mqtt.vacuum]". Before I have checked the logs I restarted the server and the vacuum.

My MQTT Addon says:
```
1571861785: New connection from 192.168.XXX.XX on port 1883.
[INFO] found valetudo on local database
1571861786: New client connected from 192.168.XXX.XX as mqttjs_88c98dbc (p2, c1, k60, u'valetudo').
```

dennis071nl, 2019-10-23:
Do you have a mqtt entry in your co figuration.yaml?? If so: remove it!

bonze4, 2019-10-27:
@dennis071nl I don't have an Mqtt entry in my configuration.yaml

MitschiBoy, 2019-11-19:
@bonze4 Try publishing a retained message to the topic
`homeassistant/vacuum/valetudo_rockrobo/config`
with the following payload
`{
  "name": "rockrobo",
  "unique_id": "rockrobo",
  "schema": "state",
  "supported_features": [
    "start",
    "pause",
    "stop",
    "return_home",
    "battery",
    "status",
    "locate",
    "clean_spot",
    "fan_speed",
    "send_command"
  ],
  "command_topic": "valetudo/rockrobo/command",
  "state_topic": "valetudo/rockrobo/state",
  "set_fan_speed_topic": "valetudo/rockrobo/set_fan_speed",
  "fan_speed_list": [
    "min",
    "medium",
    "high",
    "max",
    "mop"
  ],
  "send_command_topic": "valetudo/rockrobo/custom_command",
  "json_attributes_topic": "valetudo/rockrobo/attributes"
}`
After that restart Home Assistant
Adding unique_id should to the trick

gargomoma, 2019-12-01:
Hey guys, same problem here, and I cannot guess what is making HA not finding my vacuum.

I'm running:
HA 0.100.1 under docker.
Mosqqitto under docker as well.
And Valetudo 0.4.0 (from DustBuilder)

I'm on the same situation as @bonze4, mqtt is already working (zigbee2mqtt & tasmota) , and cannot see anything on the logs.

Thanks =)

MitschiBoy, 2019-12-01:
MQTT auto-discovery ist not working at the moment because of missing unique_id.
Add the following to your configuration.yaml
```
vacuum:
  - platform: mqtt
    name: "rockrobo"
    schema: state
    supported_features:
      - start
      - pause
      - stop
      - return_home
      - battery
      - status
      - locate
      - clean_spot
      - fan_speed
      - send_command
    command_topic: "valetudo/rockrobo/command"
    state_topic: "valetudo/rockrobo/state"
    set_fan_speed_topic: "valetudo/rockrobo/set_fan_speed"
    fan_speed_list:
      - min
      - medium
      - high
      - max
      - mop
    send_command_topic: "valetudo/rockrobo/custom_command"
    json_attributes_topic: "valetudo/rockrobo/attributes"
```

gargomoma, 2019-12-04:
It's pretty weird, but I couldn't see my vacuum until y removed any mqtt parameters from my configuration.yaml

[Like they explain in HA forum](https://community.home-assistant.io/t/valetudo-xiaomi-vacuum-dustcloud-alternative-live-maps/66277/111)

After doing that, I sent the mqtt payload, and I started to see my vacuum. =)

Thanks!

Hypfer, 2019-12-11:
> MQTT auto-discovery ist not working at the moment because of missing unique_id.

Is there some place where this requirement is mentioned? As far as I can tell it is optional?

MitschiBoy, 2019-12-11:
I just did a test. Auto-discovery is working without unique_id if the mqtt-integration is set up in the configuration.yaml. It does not work if the mqtt-integration is set up via the user interface by config flow.
I think it's mentioned here https://github.com/home-assistant/home-assistant/pull/16943 that unique_id is necessary.

Hypfer, 2019-12-11:
` It does not work if the mqtt-integration is set up via the user interface by config flow.`

Okay that explains why I didn't notice it. Thanks for testing that!

Hypfer, 2020-04-12:
Should be fixed

closed 2020-04-12 by Hypfer