Hypfer/Valetudo #338: MQTT with IPSymcon not working

KingKahn123, 2019-10-07:
Hi,

I have activated the MQTT function of Valetudo. In my IPSymcon MQTT Server I see 3 attributes coming from Valetudo "attributes", "map data" and "state". "attributes" is regularly updated. When I start a cleaning session Valetudo is sending continously following MQTT messages:

```
BUFFER IN | 31 5D 2C 5B 31 31 39 2C 31 32 31 5D 2C 5B 31 32 30 2C 31 32 31 5D 2C 5B 31 32 31 2C 31 32 31 5D 2C 5B 31 32 32 2C 31 32 31 5D 2C 5B 31 32 33 2C 31 32 31 5D 2C 5B 31 32 34 2C 31 32 31 5D 2C 5B 31 32 35 2C 31 32 31 5D 2C 5B 31 32 36 2C 31 32 31 5D 2C 5B 31 32 37 2C 31 32 31 5D 2C 5B 31 32 38 2C 31 32 31 5D 2C 5B 31 32 39 2C 31 32 31 5D 2C 5B 31 33 30 2C 31 32 31 5D 2C 5B 31 33 31 2C 31 32 31 5D 2C 5B 31 33 32 2C 31 32 31 5D 2C 5B 31 33 33 2C 31 32 31 5D 2C 5B 31 33 34 2C 31 32 31 5D 2C 5B 31 33 35 2C 31 32 31 5D 2C 5B 31 33 36 2C 31 32 31 5D 2C 5B 31 33 37 2C 31 32 31 5D 2C 5B 31 33 38 2C 31 32 31 5D 2C 5B 31 33 39 2C 31 32 31 5D 2C 5B 31 34 30 2C 31 32 31 5D 2C 5B 31 36 36 2C 31 32 31 5D 2C 5B 31 36 37 2C 31 32 31 5D 2C 5B 31 36 38 2C 31 32 31 5D 2C 5B 31 36 39 2C 31 32 31 5D 2C 5B 31 37 30 2C 31 32 31 5D 2C 5B 31 37 31 2C 31 32 31 5D 2C 5B 31 37 32 2C 31 32 31 5D 2C 5B 31 37 33 2C 31 32 31 5D 2C 5B 31 37 34 2C 31 32 31 5D 2C 5B 31 37 35 2C 31 32 31 5D 2C 5B 31 37 36 2C 31 32 31 5D 2C 5B 31 37 37 2C 31 32 31 5D 2C 5B 31 37 38 2C 31 32 31 5D 2C 5B 31 37 39 2C 31 32 31 5D 2C 5B 31 38 30 2C 31 32 31 5D 2C 5B 31 38 31 2C 31 32 31 5D 2C 5B 31 38 32 2C 31 32 31 5D 2C 5B 31 38 33 2C 31 32 31 5D 2C 5B 31 38 34 2C 31 32 31 5D 2C 5B 31 38 35 2C 31 32 31 5D 2C 5B 31 38 36 2C 31 32 31 5D 2C 5B 31 38 37 2C 31 32 31 5D 2C 5B 31 38 38 2C 31 32 31 5D 2C 5B 31 38 39 2C 31 32 31 5D 2C 5B 31 39 30 2C 31 32 31 5D 2C 5B 31 39 31 2C 31 32 31 5D 2C 5B 31 39 32 2C 31 32 31 5D 2C 5B 31 39 33 2C 31 32 31 5D 2C 5B 31 39 34 2C 31 32 31 5D 2C 5B 31 39 35 2C 31 32 31 5D 2C 5B 31 39 36 2C 31 32 31 5D 2C 5B 31 39 37 2C 31 32 31 5D 2C 5B 31 39 38 2C 31 32 31 5D 2C 5B 31 39 39 2C 31 32 31 5D 2C 5B 32 30 30 2C 31 32 31 5D 2C 5B 32 30 31 2C 31 32 31 5D 2C 5B 32 30 32 2C 31 32 31 5D 2C 5B 32 30 33 2C 31 32 31 5D 2C 5B 32 30 34 2C 31 32 31 5D 2C 5B 32 30 35 2C 31 32 31 5D 2C 5B 32 30 36 2C 31 32 31 5D 2C 5B 32 30 37 2C 31 32 31 5D 2C 5B 32 30 38 2C 31 32 31 5D 2C 5B 32 30 39 2C 31 32 31 5D 2C 5B 32 31 30 2C 31 32 31 5D 2C 5B 32 31 31 2C 31 32 31 5D 2C 5B 32 31 32 2C 31 32 31 5D 2C 5B 32 31 33 2C 31 32 31 5D 2C 5B 32 31 34 2C 31 32 31 5D 2C 5B 32 31 35 2C 31 32 31 5D 2C 5B 32 31 37 2C 31 32 31 5D 2C 5B 32 31 38 2C 31 32 31 5D 2C 5B 32 31 39 2C 31 32 31 5D 2C 5B 32 32 30 2C 31 32 31 5D 2C 5B 32 32 31 2C 31 32 31 5D 2C 5B 32 32 32 2C 31 32 31 5D 2C 5B 31 31 30 2C 31 32 30 5D 2C 5B 31 31 31 2C 31 32 30 5D 2C 5B 31 31 32 2C 31 32 30 5D 2C 5B 31 31 33 2C 31 32 30 5D 2C 5B 31 31 34 2C 31 32 30 5D 2C 5B 31 31 35 2C 31 32 30 5D 2C 5B 31 31 36 2C 31 32 30 5D 2C 5B 31 31 37 2C 31 32 30 5D 2C 5B 31 31 38 2C 31 32 30 5D 2C 5B 31 31 39 2C 31 32 30 5D 2C 5B 31 32 30 2C 31 32 30 5D 2C 5B 31 32 31 2C 31 32 30 5D 2C 5B 31 32 32 2C 31 32 30 5D 2C 5B 31 32 33 2C 31 32 30 5D 2C 5B 31 32 34 2C 31 32 30 5D 2C 5B 31 32 35 2C 31 32 30 5D 2C 5B 31 32 36 2C 31 32 30 5D 2C 5B 31 32 37 2C 31 32 30 5D 2C 5B 31 32 38 2C 31 32 30 5D 2C 5B 31 32 39 2C 31 32 30 5D 2C 5B 31 33 30 2C 31 32 30 5D 2C 5B 31 33 31 2C 31 32 30 5D 2C 5B 31 33 32 2C 31 32 30 5D 2C 5B 31 33 33 2C 31 32 30 5D 2C 5B 31 33 34 2C 31 32 30 5D 2C 5B 31 33 35 2C 31 32 30 5D 2C 5B 31 33 36 2C 31 32 30 5D 2C 5B 31 33 37 2C 31 32 30 5D 2C 5B 31 33 38 2C 31 32 30 5D 2C 5B 31 33 39 2C 31 32 30 5D 2C 5B 31 34 30 2C 31 32 30 5D 2C 5B 31 36 36 2C 31 32 30 5D 2C 5B 31 36 37 2C 31 32 30 5D 2C 5B 31 36 38 2C 31 32 30 5D 2C 5B 31 36 39 2C 31 32 30 5D 2C 5B 31 37 30 2C 31 32 30 5D 2C 5B 31 37 31 2C 31 32 30 5D 2C 5B 31 37 32 2C 31 32 30 5D 2C 5B 31 37 33 2C 31 32 30 5D 2C 5B 31 37 34 2C 31 32 30 5D 2C 5B 31 37 35 2C 31 32 30 5D 2C 5B 31 37 36 2C 31 32 30 5D 2C 5B 31 37 37 2C 31 32 30 5D 2C 5B 31 37 38 2C 31 32 30 5D 2C 5B 31 37 39 2C 31 32 30 5D 2C 5B 31 38 30 2C 31 32 30 5D 2C 5B 31 38 31 2C 31 32 30 5D 2C 5B 31 38 32 2C 31 32 30 5D 2C 5B 31 38 33 2C 31 32 30 5D 2C 5B 31 38 34 2C 31 32 30 5D 2C 5B 31 38 35 2C 31 32 30 5D 2C 5B 31 38 36 2C 31 32 30 5D 2C 5B 31 38 37 2C 31 32 30 5D 2C 5B 31 38 38 2C 31 32 30 5D 2C 5B 31 38 39 2C 31 32 30 5D 2C 5B 31 39 30 2C 31 32 30 5D 2C 5B 31 39 31 2C 31 32 30 5D 2C 5B 31 39 32 2C 31 32 30 5D 2C 5B 31 39 33 2C 31 32 30 5D 2C 5B 31 39 34 2C 31 32 30 5D 2C 5B 31 39 35 2C 31 32 30 5D 2C 5B 31 39 36 2C 31 32 30 5D 2C 5B 31 39 37 2C 31 32 30 5D 2C 5B 31 39 38 2C 31 32 30 5D 2C 5B 31 39 39 2C 31 32 30 5D 2C 5B 32 30 30 2C 31 32 30 5D 2C 5B 32 30 31 2C 

Incomplete packet. Wait for more data
```

These continuously messages crashing my IPSymcon system. Do you have any clue what the problem with my installation is?

Hypfer, 2019-10-09:
Most likely an issue with the MQTT Broker integrated in IPSymcon. We also had issue reports from FHEM Users since their Broker was broken (as in: not standard compliant) as well.
See: https://github.com/Hypfer/Valetudo/commit/28149f098fe13930c48611b74373210719027ba4

The bytes you've posted decode to a chunk of comma-separated arrays of two numbers each.
That's the map data format you're seeing there. Pixel-coordinates to be precise

My assumption is that the broker can't handle properly large MQTT messages. They most likely didn't add those as a test case because afaik they don't appear often in the wild :)

You might want to report this to Symcon GmbH. According to the mqtt spec, the maximum payload size is 256mb and this is _far_ below that limit.

KingKahn123, 2019-10-12:
Ok, the issue is solved. Not quite sure what the reason was, but now it is working without crashing my system...

closed 2019-10-26 by Hypfer