Hypfer/Valetudo #341: Enhance "No-map" FAQ - question

dustmon, 2019-10-09:
The faq says:
>Since v0.3.0 Valetudo now use the cloud interface and that requires the robot to be provisioned (wifi configured). Therefore, the map will not be displayed in AP mode! Ensure you added your device to your own wifi network. In AP mode, a map will nevertheless be created, that map can later be displayed once connected to the wifi network.

My question is: why does the bot need to be provisioned? Can you give more details about the reasons behind that? 
Especially what do you mean by "Valetudo now use the cloud interface". Which cloud interface does it use and why does it not work locally on the robot? Is an internet connection also required? The project says following in it's readme so I'm a bit confused:
> Valetudo is a standalone binary which runs on rooted Xiaomi Vacuums and aims to enable the user to operate the robot vacuum **without any Cloud Connection** whatsoever.

Thanks in advance

rand256, 2019-10-12:
Before 0.3.0 valetudo parsed local files to get map data from. This led to numerous issues, instability, different lags etc. After 0.3.0 valetudo itself pretends to be a "cloud" for a device. A "cloud" running locally on the robot. This way misguided device sends prepared map data to the valetudo (since it thinks valetudo is the cloud, remember?)

You may ask why then it requires the robot to be provisioned? It is because the roborock software logically supposes that when the robot is unprovisioned it is meaningless to use so-called "cloud connection", since an unprovisioned device by definition equals no internet connection. And, as everybody understands, normal cloud can't by definition be connected without it. So it doesn't even try.

And then you may ask, it is possible to convince the device to try using cloud connection interface in unprovisioned state? For now the answer is no. Nobody has succeeded in this task. 

Hope this helps.

dustmon, 2019-10-12:
Hi @rand256,

thanks so much for the response. I read through the slides from the author's of dustcloud and found out that the AppProxy binary is responsible for uploading the map data. 

So the goal would be to make AppProxy send the map commands regardless if the vacuum is provisioned or not. As far as I could see AppProxy is still a black box. Maybe with some reverse engineering one could find out how AppProxy determinates if the vacuum is provisioned or not. 

But this goes beyond scope of this issue.

closed 2019-10-12 by dustmon

noscript, 2021-02-10:
It's been a while, any luck with patching AppProxy or somehow faking the provisioned mode?

Hypfer, 2021-02-10:
What would be the use-case that requires this? @noscript

noscript, 2021-02-10:
To prevent access to outside world. I would like the robot be not connected anywhere by default and act as a standalone device. 

Having WiFi in AP mode allows to connect to the robot directly without anything in the middle, but then no map data is available.

Hypfer, 2021-02-10:
Keep in mind that many IoT Devices don't feature a RTC so they at least need NTP connectivity if you're planning on doing something time-based (e.g. schedules) with them