Hypfer/Valetudo #343: Support for Xiaomi Mi Robot 1s?

taucher4000, 2019-10-11:
Hi,

I just bought a Xiaomi Mi Robot 1s which is the latest model of xiaomi robots and I'm wonder if Valetudo also supports this model already or if there is any plan to support this model in the future?

Thanks,
Taucher

rand256, 2019-10-12:
You'd better wonder whether you robot could be hacked to get filesystem access to it at the first place. Hint: no.

Unless this changes, there will be no support for anything in those latest models.

tenkaibr, 2019-10-22:
Is the roborock s50 gen1 or gen2?

closed 2019-10-26 by Hypfer

argonym, 2020-06-17:
Related: https://github.com/dgiese/dustcloud/issues/252

Hypfer, 2020-06-17:
It _should_ work with 0.5.3

I haven't tested it though. Also, you need to of course acquire root which is hard on these things