Hypfer/Valetudo #344: Cannot edit zone

TheLastProject, 2019-10-12:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Screenshots**

**Vacuum Model:** S55

**Valetudo Version: ** 0.4.0

**User-Agent** Mozilla/5.0 (X11; Fedora; Linux x86_64; rv:69.0) Gecko/20100101 Firefox/69.0

**Additional context**
I never had any zones before.

lsiudut, 2019-10-12:
Same issue, here's console output:
```
Uncaught (in promise) TypeError: e._getToolbarLeftItemsElement is not a function
    at e.value (onsenui.min.js:2)
    at onsenui.min.js:2
    at onsenui.min.js:2
    at Array.forEach (<anonymous>)
    at e (onsenui.min.js:2)
    at n (onsenui.min.js:2)
    at e.value (onsenui.min.js:2)
    at onsenui.min.js:2
    at new Promise (<anonymous>)
    at onsenui.min.js:2
value @ onsenui.min.js:2
(anonymous) @ onsenui.min.js:2
(anonymous) @ onsenui.min.js:2
e @ onsenui.min.js:2
n @ onsenui.min.js:2
value @ onsenui.min.js:2
(anonymous) @ onsenui.min.js:2
(anonymous) @ onsenui.min.js:2
Promise.catch (async)
value @ onsenui.min.js:2
value @ onsenui.min.js:2
window.fn.pushPage @ (index):38
(anonymous) @ VM228:16
request.onload @ (index):57
load (async)Uncaught (in promise) TypeError: e._getToolbarLeftItemsElement is not a function
    at e.value (onsenui.min.js:2)
    at onsenui.min.js:2
    at onsenui.min.js:2
    at Array.forEach (<anonymous>)
    at e (onsenui.min.js:2)
    at n (onsenui.min.js:2)
    at e.value (onsenui.min.js:2)
    at onsenui.min.js:2
    at new Promise (<anonymous>)
    at onsenui.min.js:2
value @ onsenui.min.js:2
(anonymous) @ onsenui.min.js:2
(anonymous) @ onsenui.min.js:2
e @ onsenui.min.js:2
n @ onsenui.min.js:2
value @ onsenui.min.js:2
(anonymous) @ onsenui.min.js:2
(anonymous) @ onsenui.min.js:2
Promise.catch (async)
value @ onsenui.min.js:2
value @ onsenui.min.js:2
window.fn.pushPage @ (index):38
(anonymous) @ VM228:16
request.onload @ (index):57
load (async)
window.fn.request @ (index):55
switchToMapZoneEdit @ VM228:10
onclick @ (index):1
window.fn.request @ (index):55
switchToMapZoneEdit @ VM228:10
onclick @ (index):1
```

Athemis, 2019-10-18:
Same issue here, using Valetudo 0.4.0

ElePhontitis, 2019-10-25:
Same Issue here, also Valetudo 0.4.0 
Browser: Chromium Version 76.0.3809.132 (Official Build) Arch Linux (64-bit)

ElePhontitis, 2019-10-25:
Update: Works with Vanilla Firefox(No Plugins) and in Chrome/Chromium Incognito mode. 

Also, it seems to be a browser extension related problem with OnsenUI. Some extensions seem to overwrite critical js functions in a dirty way, which causes the zone view to crash. 
On my machine I could pin down the crashes to this chromium extension:

 "Plasma Integration 1.6.1"

By deactivating my browser extensions one by one.

The Issue https://github.com/OnsenUI/OnsenUI/issues/2211 also points out, that the plugin "LastPass" can cause the problem:

> LastPass dirty overrides document.createElement by saving current reference (at the moment the page just start loading) to native method as base method inside a closure, but a moment later CustomElements library replaces Document.prototype.createElement. But it is never called, because document.createElement references to LastPass overridden method that call saved initial method inside.

I suggest to close this issue, as it is out of Valetudos scope to fix this.

Hypfer, 2019-10-26:
@ElePhontitis thanks for looking into this :heart:

closed 2019-10-26 by Hypfer