Hypfer/Valetudo #345: Discovering

sukoshi1507, 2019-10-15:
Hello,

I just discovered the project but as I see the warning regarding the 'do not upgrade to 4004' I also realize that my two gen2 were upgraded.

I'm not able to find a reason. Changes in the firmware which makes them unrootable ?

Details would be appreciated for newcomers like me :)

rand256, 2019-10-15:
Changes in the firmware make it impossible to locally flash another firmware image when you'll decide to upgrade from that one. Now flashing procedure allows downloading images only from remote xiaomi servers.

sukoshi1507, 2019-10-15:
So i'm totally screwed now.

Thanks :)

closed 2019-10-15 by sukoshi1507

rand256, 2019-10-15:
You can simply do factory reset to return to the upgradable initial firmware.

Also I think it should be possible to initiate firmware upgrade process by sending needed commands from the cloud interface, which valetudo emulates since 0.3.0. Just nobody implemented that yet.

sukoshi1507, 2019-10-16:
I'll be the guinea pig if needed :-)