Hypfer/Valetudo #347: Cant reach webserver to enter wifi credentials

wAuner, 2019-10-16:
I have the roborock second generation. I've reset the wifi settings and followed the instructions from the tutorial and I've built my image with dustbuilder. Everythig worked fine but when I finally want to connect to the webserver on `192.168.8.1` while I'm connected to the robots wifi, nothing happens, the site can't be reached. I'm able to connect via ssh though.

Any ideas what went wrong? Thanks!

Jackfritt, 2019-10-16:
Same problem here :)

nagel86, 2019-10-18:
This might help:
For reasons yet to be discovered, the Valetudo service might not be started on a newly flashed robot. You may need to connect via SSH and run `service valetudo start`

Jackfritt, 2019-10-18:
Ok found my problem. I didnt compile  Valetudo from git
git clone https://github.com/Hypfer/Valetudo.git
cd Valetudo/
npm install
npm run build

This was the difference from the Installation Instructions. There you should 
download the bin file from Valetudo. I wanted the latest ang greatest stuff :)

Thx for the hints.

closed 2020-01-15 by Hypfer