Hypfer/Valetudo #348: Web-App does not refresh

aimtrainer, 2019-10-20:
**Describe the bug**

Chrome on android has the option "Add to Home screen". When adding Valetudo to the homescreen and afterwards starting it via the generated shortcut, it is opened as a Web-App, meaning there are no chrome elements visable. When opening Valetudo like this the for the first time it works as expected. When opening it again later, it does not refresh the displayed information. Swiping down does not refresh the page either (as it would normally in chrome) and there's no refresh button either.

**To Reproduce**

Add to Home screen as described above, view info under Home, start cleanup and leave Valetudo. When viewing the Web-App after the Cleanup it does not show the new info about the cleaned area and the cleaning time.

I've also encountered the problem, that I changed the "Persistent Data" setting in the Web-App and after saving it did not show the changed setting in the Web-App (it did when viewing it normally via browser).

**Expected behavior**

When the Web-App is open it should either refresh regularly or at least give the user the opportunity to refresh manually.

**Vacuum Model:** 
RoboRock S50 (Firmware 3.3.9)
**Valetudo Version: ** 
0.4.0
**User-Agent** 
Android Chrome / Web-App

waclaw66, 2019-10-27:
Seems like websocket issue. Have you tried to kill Chrome before you start shortcut again?

aimtrainer, 2019-10-27:
Hi Václav,

yea that works, but it's not really a satisfying solution.

Regards Felix

ottelo9, 2019-11-05:
I have problems with iOS (iPhone X) and the map not showing if I use the http password login. The map is always grey and a blue loading bard appears on the top of the screen.

Hypfer, 2021-01-12:
Since pretty much the whole codebase changed recently, please re-check if this is still an issue in [the latest beta](https://github.com/Hypfer/Valetudo/releases/tag/2021.01.0b0) and if yes open a new issue.

Thank you!

closed 2021-01-12 by Hypfer