Hypfer/Valetudo #353: Map only showing "no map data"

Bragorr, 2019-10-25:
I updated my Xiaomi Mi Gen. 1 to valetudo 0.4.0 with Firmware v11_003468. After that the map displays only what I assume the default map that paints "NO_MAP_DATA". I rebooted - didn't help. I did a factory reset - didn't help. I looked into /var/log/upstart/valetudo.log no error message but also no "Robot connected" message, only:
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Help would be very appreciated!

**To Reproduce**
1. Open valetudo webpage
2. Click on map
3. See "NO_MAP_DATA"-image

**Expected behavior**
1. Open valetudo webpage
2. Click on Map
3. See map of the room(s)

**Screenshots**

**Vacuum Model:** 
Gen1
**Valetudo Version: ** 
0.4.0
**User-Agent** 
Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:69.0) Gecko/20100101 Firefox/69.0
Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/78.0.3904.70 Safari/537.36
**Additional context**

ElePhontitis, 2019-10-25:
Is your Robot provisioned(connected to your wifi)? 

Did you load the binary from the release assets, or did you build it yourself? The 0.4.0 release asset binary somehow still has version 0.3.1 in it, so building it yourself is better.

What is the content of the files `/etc/hosts` and `/etc/rc.local` on your robot?

Bragorr, 2019-10-25:
Yes, my robot is connected to my wifi.
I did load the binary from the release assets. But the robot shows on the device information webpage that the valetudo version is 0.4.0.
/etc/hosts:
```
127.0.0.1       localhost
127.0.1.1       rockrobo

::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

0.0.0.0       awsbj0-files.fds.api.xiaomi.com
0.0.0.0       awsbj0.fds.api.xiaomi.com

```

/etc/rc.local:
```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

exit 0

```

ElePhontitis, 2019-10-25:
You are missing entries in both files. 
Can you add the content of` valetudo/deployment/etc/hosts` in your flashing directory to `/etc/hosts` on your robot? The same for `valetudo/deployment/etc/rc.local` and `/etc/rc.local`(put the entries before `exit 0`)

After that reboot your robot

Bragorr, 2019-10-25:
I have no idea why these lines were not in the files but after adding them the map works like a charm.
You are my hero! Thanks a lot!

ElePhontitis, 2019-10-26:
Happy to help!

closed 2019-10-26 by Hypfer

peterge1998, 2020-02-16:
> You are missing entries in both files.
> Can you add the content of` valetudo/deployment/etc/hosts` in your flashing directory to `/etc/hosts` on your robot? The same for `valetudo/deployment/etc/rc.local` and `/etc/rc.local`(put the entries before `exit 0`)
> 
> After that reboot your robot

Can confirm this worked for me too!

https://raw.githubusercontent.com/Hypfer/Valetudo/master/deployment/etc/rc.local
https://github.com/Hypfer/Valetudo/raw/master/deployment/etc/hosts

Horstbart, 2020-05-26:
I had the same issue showing "no map data" on Release 0.5.0 .  @ElePhontitis 's recommended steps solved the problem for me as well. Many thanks !