Hypfer/Valetudo #356: Map data traffic

waclaw66, 2019-10-28:
**Describe the bug**
Showing the live map, the data traffic is about 300kB/s. It's too much for mobile data.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Map' tab

**Expected behavior**
Data traffic should be less.

**Vacuum Model:** Gen1

**Valetudo Version:** 0.4.0

**User-Agent:** Mozilla/5.0 (Linux; Android 9; Mi 9T) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/77.0.3865.116 Mobile Safari/537.36

**Additional context**

x10an14, 2019-10-30:
Would you mind sharing how you're measuring this?

E.g. what tools/technology/commands do you use to profile this?

waclaw66, 2019-10-30:
> 
> 
> Would you mind sharing how you're measuring this?
> 
> E.g. what tools/technology/commands do you use to profile this?

First, I noticed high traffic in MIUI status bar while displaying the map. That led me to suspicion, then I observed the connection traffic using jnettop command on the router.

Hypfer, 2020-01-15:
Yes but how

closed 2020-01-15 by Hypfer

waclaw66, 2020-01-15:
> 
> 
> Yes but how

Very simply, compress websocket json messages. Each map message has over 260kB in my case and is sent every second when map tab is activated.

Hypfer, 2020-01-15:
https://github.com/Hypfer/Valetudo/blob/9fc8db27d757b196b3036028468ec5f905d216f8/lib/webserver/WebServer.js#L935

That's already happening

waclaw66, 2020-01-15:
> 
> 
> https://github.com/Hypfer/Valetudo/blob/9fc8db27d757b196b3036028468ec5f905d216f8/lib/webserver/WebServer.js#L935
> 
> That's already happening

Didn't know that. Then the question is why a browser (tested on Firefox, Chrome on Windows and Android) receives huge uncompressed json messages when they're compressed on server side on vacuum cleaner :/

Hypfer, 2020-01-15:
The code was open all along so there's no real reason not to check it first.

There are at least four possibilities that come to my mind

1. The browser transparently inflates the data and it just looks like it's not compressed
2. There's a reverse proxy inbetween which transparently inflates the data
3. Old version of valetudo

4. It's actually broken

waclaw66, 2020-01-15:
Ad 1) I have monitored network traffic on Mikrotik router and NetLimiter firewall. The same results, download about 400kB/s.
Ad 2) That was my suspicion, i use Reverse proxy on Apache. Unfortunately the I got the same results while connecting directly to IP address. I have also checked traffic from vacuum cleaner in Wireshark, websocket messages are really uncompressed.
Ad 3) Using the latest 0.4.0 version of Valetudo.
Ad 4) Maybe, more debug needed. Can you help, what to check?

Thanks!

waclaw66, 2020-01-15:
I noticed that compression is not present within the latest release 0.4.0. It has been added with https://github.com/mkiolokolo/Valetudo/pull/46

waclaw66, 2020-01-22:
Valetudo consumed all my mobile data FUP today, almoast 3GB :( That's shame, that there in no new release with compression until today. I've tried a fork Valetudo RE, data consumtion about 4kB/s, more features. Sorry to say that, I'm switching...