Hypfer/Valetudo #357: no map data

Nyrk2k, 2019-10-30:
**Describe the bug**

I started many cleanups, but valetudo has no map data anymore.

The Config:
`
{
  "spots": [],
  "areas": [],
  "mqtt": {
    "enabled": false,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "mqtt://user:pass@foobar.example",
    "provideMapData": true,
    "caPath": ""
  },
  "dummycloud": {
    "spoofedIP": "203.0.113.1",
    "bindIP": "127.0.0.1"
  },
  "httpAuth": {
    "enabled": false,
    "username": "valetudo",
    "password": "valetudo"
  },
  "allowSSHKeyUpload": true,
  "map_upload_host": "http://127.0.0.1"
}`

map upload is set to 127.0.0.1 but no data is set.

the map dir is empty

**To Reproduce**

**Expected behavior**

**Screenshots**

**Vacuum Model:** 

**Valetudo Version: ** 
0.4.0

**User-Agent** 

**Additional context**

loswochos, 2019-10-31:
Have the same issue since a few month and valetudo 0.3.0. Built a new image today based on FW v11_001886 with valetudo 0.4.0. Followed the installation instructions, checked all configs, no success. Don't know whats wrong. Any hints?

bartekd123, 2019-10-31:
Weird. Mine was working fine last 2 weeks, but don’t have nap data today either. I’ll see tomorrow when it vacuums to see if it comes back.

ElePhontitis, 2019-11-01:
Can you try the instructions I added to this issue https://github.com/Hypfer/Valetudo/issues/353?

bartekd123, 2019-11-02:
Actually I didn’t do anything and today the map data showed up. I tried doing a clean up yesterday and the map data didn’t generate. Could it have anything to do with the date October 31st?

peritus, 2019-11-07:
I'm experiencing the same problem (fresh install of 0.4.0 via the build service yesterday). I checked the two files mentioned in #353 and the files on the vacuum match the files in this repository.

onggie, 2019-11-27:
Anyone have clarity on this, how have the maps all of a sudden stopped working?

Fresh install of Valetudo, multiple reflashes of the device. This was working, now it's not.

simlur, 2019-12-03:
My map was missing with FW v11_001886 and valetudo 0.4.0 fetched from Dustbuilder yesterdy. Turning Persistent Data off and on and than restarting the system brought the map back. 

Could see a map after factory reset + flashing + a few seconds of cleaning. It disapeared when starting cleaning again. Found that Persistent Data had turned off.

onggie, 2019-12-03:
> 
> Could see a map after factory reset + flashing + a few seconds of cleaning. It disapeared when starting cleaning again. Found that Persistent Data had turned off.

Gen 1 doesn't have persistent maps and still the maps persist as long as you don't initiate a full clean.

Lockzi, 2019-12-07:
I just figured something out on my device having the exact same problem.

Did you configure WiFi through imagebuilder.sh's: 
       --unprovisioned=wpa2psk --ssid=BestWiFiEver --psk=abcd1234
?

I did, and aparanetly I had to do it through Valetudo web interface for the map to show up!

onggie, 2019-12-07:
> I just figured something out on my device having the exact same problem.
> 
> Did you configure WiFi through imagebuilder.sh's:
> --unprovisioned=wpa2psk --ssid=BestWiFiEver --psk=abcd1234
> ?
> 
> I did, and aparanetly I had to do it through Valetudo web interface for the map to show up!

Nope, I did it through Valetudo.

onggie, 2019-12-13:
Some more developments on this, I realized my rc.local and hosts was empty after manually adding these in.

Re entering the values and performing chattr +i lost me access to the robot.

hosts look like this

> 127.0.0.1       localhost
> 127.0.1.1       rockrobo
> 
> ::1             localhost ip6-localhost ip6-loopback
> ff02::1         ip6-allnodes
> ff02::2         ip6-allrouters
> 
> ### VALETUDO HOSTS INIT ###
> ###########################################################
> ### VALETUDO HOSTS
> ###########################################################
> #
> # Add to /etc/hosts
> #
> 
> 203.0.113.1  ot.io.mi.com
> 2001:db8::1  ot.io.mi.com
> 203.0.113.1  ott.io.mi.com
> 2001:db8::1  ott.io.mi.com
> 203.0.113.1  us.ot.io.mi.com
> 2001:db8::1  us.ot.io.mi.com
> 203.0.113.1  us.ott.io.mi.com
> 2001:db8::1  us.ott.io.mi.com
> 203.0.113.1  sg.ot.io.mi.com
> 2001:db8::1  sg.ot.io.mi.com
> 203.0.113.1  sg.ott.io.mi.com
> 2001:db8::1  sg.ott.io.mi.com
> 203.0.113.1  st.ot.io.mi.com
> 2001:db8::1  st.ot.io.mi.com
> 203.0.113.1  st.ott.io.mi.com
> 2001:db8::1  st.ott.io.mi.com
> 203.0.113.1  de.ot.io.mi.com
> 2001:db8::1  de.ot.io.mi.com
> 203.0.113.1  de.ott.io.mi.com
> 2001:db8::1  de.ott.io.mi.com
> 203.0.113.1  ru.ot.io.mi.com
> 2001:db8::1  ru.ot.io.mi.com
> 203.0.113.1  ru.ott.io.mi.com
> 2001:db8::1  ru.ott.io.mi.com
> 203.0.113.1  in.ot.io.mi.com
> 2001:db8::1  in.ot.io.mi.com
> 203.0.113.1  in.ott.io.mi.com
> 2001:db8::1  in.ott.io.mi.com
> 203.0.113.1  tw.ot.io.mi.com
> 2001:db8::1  tw.ot.io.mi.com
> 203.0.113.1  tw.ott.io.mi.com
> 2001:db8::1  tw.ott.io.mi.com
> 203.0.113.1  ea.ot.io.mi.com
> 2001:db8::1  ea.ot.io.mi.com
> 203.0.113.1  ea.ott.io.mi.com
> 2001:db8::1  ea.ott.io.mi.com
> 203.0.113.1  pv.ot.io.mi.com
> 2001:db8::1  pv.ot.io.mi.com
> 203.0.113.1  pv.ott.io.mi.com
> 2001:db8::1  pv.ott.io.mi.com
> 203.0.113.1  cdn.cnbj0.files.fds.api.xiaomi.com
> 2001:db8::1  cdn.cnbj0.files.fds.api.xiaomi.com
> 203.0.113.1  cnbj0.fds.api.xiaomi.com
> 2001:db8::1  cnbj0.fds.api.xiaomi.com
> 203.0.113.1  cnbj2.fds.api.xiaomi.com
> 2001:db8::1  cnbj2.fds.api.xiaomi.com
> 203.0.113.1  awsbj0.fds.api.xiaomi.com
> 2001:db8::1  awsbj0.fds.api.xiaomi.com
> 203.0.113.1  awsbj0-files.fds.api.xiaomi.com
> 2001:db8::1  awsbj0-files.fds.api.xiaomi.com
> 203.0.113.1  awsusor0.fds.api.xiaomi.com
> 2001:db8::1  awsusor0.fds.api.xiaomi.com
> 203.0.113.1  awsusor0-files.fds.api.xiaomi.com
> 2001:db8::1  awsusor0-files.fds.api.xiaomi.com
> 203.0.113.1  awssgp0.fds.api.xiaomi.com
> 2001:db8::1  awssgp0.fds.api.xiaomi.com
> 203.0.113.1  awssgp0-files.fds.api.xiaomi.com
> 2001:db8::1  awssgp0-files.fds.api.xiaomi.com
> 203.0.113.1  awsde0.fds.api.xiaomi.com
> 2001:db8::1  awsde0.fds.api.xiaomi.com
> 203.0.113.1  awsde0-files.fds.api.xiaomi.com
> 2001:db8::1  awsde0-files.fds.api.xiaomi.com
> 203.0.113.1  ksyru0-eco.fds.api.xiaomi.com
> 2001:db8::1  ksyru0-eco.fds.api.xiaomi.com
> 203.0.113.1  ksyru0-eco-files.fds.api.xiaomi.com
> 2001:db8::1  ksyru0-eco-files.fds.api.xiaomi.com
> 203.0.113.1  awsind0-eco.fds.api.xiaomi.com
> 2001:db8::1  awsind0-eco.fds.api.xiaomi.com
> 203.0.113.1  awsind0-eco-files.fds.api.xiaomi.com
> 2001:db8::1  awsind0-eco-files.fds.api.xiaomi.com
> ### VALETUDO HOSTS EXIT ### 

rc.local looks like this

> #!/bin/sh -e
> #
> # rc.local
> #
> # This script is executed at the end of each multiuser runlevel.
> # Make sure that the script will "exit 0" on success or any other
> # value on error.
> #
> # In order to enable or disable this script just change the execution
> # bits.
> #
> # By default this script does nothing.
> iptables         -F OUTPUT
> ip6tables        -F OUTPUT
> iptables  -t nat -F OUTPUT
> iptables  -t nat -A OUTPUT -p tcp --dport 80   -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053
> iptables  -t nat -A OUTPUT -p udp --dport 8053 -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053
> iptables         -A OUTPUT                     -d 203.0.113.1/32  -j REJECT
> ip6tables        -A OUTPUT                     -d 2001:db8::1/128 -j REJECT
> 
> exit 0

Hypfer, 2020-01-15:
Since there a a _lot_ of working installations, there must be something wrong with your setup.

closed 2020-01-15 by Hypfer