Hypfer/Valetudo #358: Unable to start Valetudo binary on roborock S55 (Gen2)

x10an14, 2019-10-31:
**Describe the bug**
Unable to start `valetudo` binary on roborock.
(It has started as late as yesterday).

**To Reproduce**

I can faithfully reproduce it like this on my roborock;
```bash
[2019-10-31 22:14:26] 0 x10an14@x10-desktop:~
-> $ ssh roborock 
Enter passphrase for key '/home/x10an14/.ssh/id_rsa': 
Welcome to Ubuntu 14.04.6 LTS (GNU/Linux 3.4.39 armv7l)

 * Documentation:  https://help.ubuntu.com/
Last login: Thu Oct 31 22:11:03 2019 from x10-desktop.x10network
root@x10roborock:~# /usr/local/bin/valetudo 
Pkg: Error reading from file.
root@x10roborock:~# 
```

**Expected behavior**
Valetudo webserver should start.

Either by `service valetudo (re)start`, or by just executing the `valetudo` binary on `$PATH` (which in my case - and I guess most/everybody else - is `/usr/lib/local/valetudo`).

**Vacuum Model:**
Xiaomi Roborock S55 (Gen2 I guess that is).

**Valetudo Version:**
`0.4.0`

**User-Agent**
N/A.

**Additional context**
No clue what that would be at this point.

I've run `strace /usr/lib/local/valetudo`.
If that's of any help, below is the output of said `strace`.

<details>
<summary><code>strace</code> output</summary>

```bash
[2019-10-31 22:14:26] 0 x10an14@x10-desktop:~
-> $ ssh roborock 
Enter passphrase for key '/home/x10an14/.ssh/id_rsa': 
Welcome to Ubuntu 14.04.6 LTS (GNU/Linux 3.4.39 armv7l)

 * Documentation:  https://help.ubuntu.com/
Last login: Thu Oct 31 22:11:03 2019 from x10-desktop.x10network
root@x10roborock:~# /usr/local/bin/valetudo 
Pkg: Error reading from file.
root@x10roborock:~# strace /usr/local/bin/valetudo 
execve("/usr/local/bin/valetudo", ["/usr/local/bin/valetudo"], [/* 33 vars */]) = 0
brk(0)                                  = 0x19ac000
uname({sys="Linux", node="x10roborock", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=20678, ...}) = 0
mmap2(NULL, 20678, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f96000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\240\t\0\0004\0\0\0"..., 512) = 512
lseek(3, 8664, SEEK_SET)                = 8664
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1160) = 1160
lseek(3, 8328, SEEK_SET)                = 8328
read(3, "A4\0\0\0aeabi\0\1*\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 53) = 53
[... 2414 lines cut ...]
clock_gettime(CLOCK_MONOTONIC, {2847, 292245068}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 292717318}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 293303609}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 294140068}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 294943984}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 295428484}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 295862359}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 296331151}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 297138901}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 299508651}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 300964859}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 301445859}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 301970568}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 302395901}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 302932568}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 304121484}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 304627859}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 304957942}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 305322817}) = 0
clock_gettime(CLOCK_MONOTONIC, {2847, 305728984}) = 0
exit_group(1)                           = ?
+++ exited with 1 +++
root@x10roborock:~# 
```
</details>

x10an14, 2019-10-31:
I guess it's worth mentioning that I did perform a `apt update && apt upgrade && apt install vim strace binutils unattended-upgrades`.

Pretty sure those were all I installed, but not 100%.
Can provide full list if relevant.

I moved & symlinked largest folders using up space on `/` over to `/mnt/data/<new folder>`.
Folders like `/var/cache/apt` and `/var/lib/apt`...

Hypfer, 2019-11-01:
Did you strip the binary? Checked the md5sum?

x10an14, 2019-11-01:
> Did you strip the binary? Checked the md5sum?
Actually, yes.

I modified the default Json config-string (inside of the binary) from specifying `Europe/Berlin` to `Europe/Oslo`.
Since I didn't get Valetudo to pick up on that when I edited `/etc/timezone`.

```bash
[2019-11-01 21:54:32] 0 x10an14@x10-desktop:~
-> $ ssh roborock 
Enter passphrase for key '/home/x10an14/.ssh/id_rsa': 
Welcome to Ubuntu 14.04.6 LTS (GNU/Linux 3.4.39 armv7l)

 * Documentation:  https://help.ubuntu.com/
Last login: Fri Nov  1 15:34:46 2019 from x10-desktop.x10network
root@x10roborock:~# which valetudo
/usr/local/bin/valetudo
root@x10roborock:~# md5sum which valetudo)
Display all 982 possibilities? (y or n)
root@x10roborock:~# md5sum $(which valetudo)
105b8e12fde3efef684cbf431a0e19ce  /usr/local/bin/valetudo
root@x10roborock:~# 
```

Would that affect this??

Hypfer, 2019-11-01:
It wouldn't surprise me if editing the binary with a hex editor would break it.

The whole package is quite fragile already

x10an14, 2019-11-01:
I just edited with `vim`, but I could try a transplant/replace of a compiled binary I guess - see if that fixes it.

x10an14, 2019-11-01:
So a download of [this](https://github.com/Hypfer/Valetudo/releases/download/0.4.0/valetudo) binary fixed it.
```bash
-> $ sha256sum valetudo 
e8c35553c81e819aa87d563426ee2019e01371980695e520f7098e72fa97faa4  valetudo
```

So my bad for editing the default json config-map I guess.

x10an14, 2019-11-01:
Closing this issue since it's basically my fault for trying to fix #359 in an erroneous manner =)

closed 2019-11-01 by x10an14