Hypfer/Valetudo #362: Can't login to SSH using root

CanuckianOz, 2019-11-02:
Xiaomi Roborock v1, running 3468 firmware. 

I had a multitude of issues getting Valetudo working as the installation instructions now seem out of date, but finally got the web service running. It's controllable using the token but not MQTT so I believe I need to turn that on in the config file on the robot. I also can't use the Hassio addon to add the map to Home Assistant. However, I can't access SSH to change the config file as it's rejecting passwords. Is it a device-specific password?

christianh17, 2019-11-03:
Hi! I think it is a problem with the username and / or the ssh key. How do you try to login? For me it works with "ssh -l root <IP of device>" after I confirmed once to trust the device. 
Do you remember how you created the ssh key?

CanuckianOz, 2019-11-04:
I've tried two ways and neither worked. I created my own ssh key and firmware using a linux machine and that uploaded fine to the robot but neither the web service nor ssh worked. Then I tried using the dustbuilder method (with them creating an ssh key) and I finally got the web service working, but SSH rejects any passwords. 

Part of my problem is I don't really understand how ssh keys work and how they relate to the machine account password that they're created on, how they get linked to ssh connections etc I don't really understand public vs private keys either.

bbarbisch, 2019-11-05:
This might be because your ssh command does not use the created certificate but instead tries to login using classic username/password authentication.

In your terminal you might see something like the following:
```
aaa@aaa:~$ ssh -l root x.x.x.x
root@x.x.x.x's password: 
```
This password dialog is displayed when username/password authentication is used.

You need to explicitly use your generated keyfile (the private key) with the ssh command:
```
aaa@aaa:~$ ssh -i <path-to-your-private-keyfile> -l root x.x.x.x
Enter passphrase for key '<path-to-your-private-keyfile>': 
```

It seems that sometimes the ssh command does not recognize keyfiles in .ssh. Therefore you have either set it explicitly using the -i option or configure a ssh config file.

CanuckianOz, 2019-11-06:
I think I found the problem but the solution is to re-flash, and I really don't want to as it took me numerous tries to get it to work this far. Unless there's an easier way once it's flashed the first time? 

I used Dustbuilder here to build the firmware file and selected "let DustBuilder generate an SSH key for you". I did this because I tried numerous times to flash it by building my own, and it was "successful" but I could never access the webserver.
https://dustbuilder.xvm.mit.edu/

Here it basically says they don't support ssh logins. 

> We do not support password logins. If you do not intend to ever connect over SSH, feel free to select "Let DustBuilder generate a SSH Keypair", e.g. if you use Valetudo.

So I think I'm shit out of luck. I either need to re-flash or give up.

CanuckianOz, 2019-11-06:
@bbarbisch forget EVERYTHING I just said. Thank you - your clue regarding specifying the SSH key helped me read to understand the DustBuilder help section better.

From my understanding, they build the firmware and embed their own ssh key, then provide that. I need to specify that ssh key to match and connect to the vacuum. I think I understand this correctly?

bbarbisch, 2019-11-06:
Yes. If you do not provide a public key, Dustbuilder will create a new keypair and embed the public key in the generated image. Afterwards it sends them to the provided email address.

To connect via ssh to your vacuum you need to use the private key (which was sent to you by email).

fyi:
- the file containing the public key ends with .pub
- the file containing the private key usually has not file extension
- alternatively you can open the file in a editor (or print it out via cat). The file with the private key contains the following header:
`-----BEGIN OPENSSH PRIVATE KEY-----
`

closed 2019-11-09 by CanuckianOz

lairdman, 2022-01-10:
@miguelpruivo Gone. Second last bug fix in https://newreleases.io/project/github/Hypfer/Valetudo/release/2021.09.1 

I lost my old ssh-key so I think I'm now locked out of my robot.

asdasddffgthhjtreresf, 2022-01-19:
> @miguelpruivo Gone. Second last bug fix in https://newreleases.io/project/github/Hypfer/Valetudo/release/2021.09.1
> 
> I lost my old ssh-key so I think I'm now locked out of my robot.

omfg. In what the fucking piece of shit have I put my hands on. This is a fucking clusterfuck.

lairdman, 2022-01-19:
Unlikely the change was made just to annoy you. It caught me out and I agree it's a minor pain, but we're getting a lot of free value out of this guys work so I'm overall pretty happy.

Reflash it, save the keys and move on.

Hypfer, 2022-01-20:
Nope, I've specificially removed that feature because I like seeing users such as `asdasddffgthhjtreresf ` suffer.
If you have additional ideas on how I could make your life worse please tell me. You deserve it.