Hypfer/Valetudo #364: WebUi not showing, service not listed, but cleaning still works

ministryofsillywalks, 2019-11-03:
**Describe the bug**

My Gen1 is not showing the webui anymore. Cleaning still works using the hardware buttons but otherwise I can't access it.
Connecting via ssh and starting the service gives me this:
_root@rockrobo:~# service valetudo start
valetudo start/running, process 4632_

However when checking
_root@rockrobo:~# service --status-all_

The valetudo service is not listed.

I tried reinstalling valetudo using RRCC (roborockcontrolcenter) but the problem remains

**Expected behavior**

The webui should of course be shown

**Screenshots**

**Vacuum Model:** Gen1

**Valetudo Version: ** 0.4.0

**User-Agent** Various

**Additional context**

CanuckianOz, 2019-11-09:
I just discovered the same problem. First I noticed I couldn't see the map but could still control the vacuum in hassio. Then I saw the flashing wifi light on the rockrobo. Can "see" the webui but no data populates and can't connect. Reset the wifi and rebooted the vacuum and was able to connect directly and reconfigure wifi for my network. Logged in via ssh and noticed there's no valetudo service anymore.

onggie, 2019-11-12:
Went to post a new issue just then.  

Here is what I'm experiencing;

- Web browsing not working
- SSH not responding
- Manually pressing the clean button on the device works

I then reset the device
- Connecting to the device with the Mi Home app works and can clean
- mirobo --debug discover --handshake true (discovers the device)
- I can flash the firmware again successfully
- Still can't SSH or web browser

None of this makes sense really, hoping someone can comment on the issue as it's weird it happen all of a sudden.

MrMooed, 2019-11-12:
> None of this makes sense really, hoping someone can comment on the issue as it's weird it happen all of a sudden.

Exact same behavior on my Gen2. It was last seen on my network on 2019-11-10 03:02:13.

onggie, 2019-11-18:
Some more testing with my limited knowledge, all Mirobo commands work on the device.  So unsure why ports 22 and 80 aren't working.  I don't have to regenerate SSH keys every time I reset the device do I?

onggie, 2019-11-19:
@MrMooed  @CanuckianOz @ministryofsillywalks 

Guys, I had a bit more luck with the following but still can't get through port 80, but at least SSH was working.

I followed the steps from the fork of Valetudo, https://github.com/rand256/valetudo

1. Got the token with Mirobo
2. Downloaded the firmware app https://forum.xda-developers.com/android/apps-games/app-xvacuum-firmware-xiaomi-vacuum-t3896526
3. Downloaded the rooted firmware https://github.com/rand256/valetudo/wiki/Firmware-downloads
4. Flashed the rooted firmware with the Android app
5. Installed Valetudo manually using the steps at the bottom (https://github.com/rand256/valetudo/wiki/Installation-process)

So everything worked fine and I could finally SSH, but my only problem is I can access via port 80.  But at least SSH works for my diagnosis.

onggie, 2019-11-19:
Sorry for the bombard of messages, putting this final one here as I got it working.  Because I used the rebuilt release of Valetudo found here https://github.com/rand256/valetudo/releases/tag/0.4.0-RE6 and I unpacked the tar.gz with 7-Zip, I forgot I needed to unzip the file within that unpacked file.

Bingo port 80 working when manually running it, should come up with a reboot now.

onggie, 2019-11-23:
Back on this, I have flashed the device using the rooted firmware from rand256, but then put on the stock version of Valetudo. Issue that appears is that it's not storing map data and just showing no map data.

onggie, 2019-12-13:
Some more developments on this, I realized my rc.local and hosts was empty after manually adding these in.

Re entering the values and performing chattr +i lost me access to the robot.

hosts look like this

> 127.0.0.1       localhost
> 127.0.1.1       rockrobo
> 
> ::1             localhost ip6-localhost ip6-loopback
> ff02::1         ip6-allnodes
> ff02::2         ip6-allrouters
> 
> ### VALETUDO HOSTS INIT ###
> ###########################################################
> ### VALETUDO HOSTS
> ###########################################################
> #
> # Add to /etc/hosts
> #
> 
> 203.0.113.1  ot.io.mi.com
> 2001:db8::1  ot.io.mi.com
> 203.0.113.1  ott.io.mi.com
> 2001:db8::1  ott.io.mi.com
> 203.0.113.1  us.ot.io.mi.com
> 2001:db8::1  us.ot.io.mi.com
> 203.0.113.1  us.ott.io.mi.com
> 2001:db8::1  us.ott.io.mi.com
> 203.0.113.1  sg.ot.io.mi.com
> 2001:db8::1  sg.ot.io.mi.com
> 203.0.113.1  sg.ott.io.mi.com
> 2001:db8::1  sg.ott.io.mi.com
> 203.0.113.1  st.ot.io.mi.com
> 2001:db8::1  st.ot.io.mi.com
> 203.0.113.1  st.ott.io.mi.com
> 2001:db8::1  st.ott.io.mi.com
> 203.0.113.1  de.ot.io.mi.com
> 2001:db8::1  de.ot.io.mi.com
> 203.0.113.1  de.ott.io.mi.com
> 2001:db8::1  de.ott.io.mi.com
> 203.0.113.1  ru.ot.io.mi.com
> 2001:db8::1  ru.ot.io.mi.com
> 203.0.113.1  ru.ott.io.mi.com
> 2001:db8::1  ru.ott.io.mi.com
> 203.0.113.1  in.ot.io.mi.com
> 2001:db8::1  in.ot.io.mi.com
> 203.0.113.1  in.ott.io.mi.com
> 2001:db8::1  in.ott.io.mi.com
> 203.0.113.1  tw.ot.io.mi.com
> 2001:db8::1  tw.ot.io.mi.com
> 203.0.113.1  tw.ott.io.mi.com
> 2001:db8::1  tw.ott.io.mi.com
> 203.0.113.1  ea.ot.io.mi.com
> 2001:db8::1  ea.ot.io.mi.com
> 203.0.113.1  ea.ott.io.mi.com
> 2001:db8::1  ea.ott.io.mi.com
> 203.0.113.1  pv.ot.io.mi.com
> 2001:db8::1  pv.ot.io.mi.com
> 203.0.113.1  pv.ott.io.mi.com
> 2001:db8::1  pv.ott.io.mi.com
> 203.0.113.1  cdn.cnbj0.files.fds.api.xiaomi.com
> 2001:db8::1  cdn.cnbj0.files.fds.api.xiaomi.com
> 203.0.113.1  cnbj0.fds.api.xiaomi.com
> 2001:db8::1  cnbj0.fds.api.xiaomi.com
> 203.0.113.1  cnbj2.fds.api.xiaomi.com
> 2001:db8::1  cnbj2.fds.api.xiaomi.com
> 203.0.113.1  awsbj0.fds.api.xiaomi.com
> 2001:db8::1  awsbj0.fds.api.xiaomi.com
> 203.0.113.1  awsbj0-files.fds.api.xiaomi.com
> 2001:db8::1  awsbj0-files.fds.api.xiaomi.com
> 203.0.113.1  awsusor0.fds.api.xiaomi.com
> 2001:db8::1  awsusor0.fds.api.xiaomi.com
> 203.0.113.1  awsusor0-files.fds.api.xiaomi.com
> 2001:db8::1  awsusor0-files.fds.api.xiaomi.com
> 203.0.113.1  awssgp0.fds.api.xiaomi.com
> 2001:db8::1  awssgp0.fds.api.xiaomi.com
> 203.0.113.1  awssgp0-files.fds.api.xiaomi.com
> 2001:db8::1  awssgp0-files.fds.api.xiaomi.com
> 203.0.113.1  awsde0.fds.api.xiaomi.com
> 2001:db8::1  awsde0.fds.api.xiaomi.com
> 203.0.113.1  awsde0-files.fds.api.xiaomi.com
> 2001:db8::1  awsde0-files.fds.api.xiaomi.com
> 203.0.113.1  ksyru0-eco.fds.api.xiaomi.com
> 2001:db8::1  ksyru0-eco.fds.api.xiaomi.com
> 203.0.113.1  ksyru0-eco-files.fds.api.xiaomi.com
> 2001:db8::1  ksyru0-eco-files.fds.api.xiaomi.com
> 203.0.113.1  awsind0-eco.fds.api.xiaomi.com
> 2001:db8::1  awsind0-eco.fds.api.xiaomi.com
> 203.0.113.1  awsind0-eco-files.fds.api.xiaomi.com
> 2001:db8::1  awsind0-eco-files.fds.api.xiaomi.com
> ### VALETUDO HOSTS EXIT ### 

rc.local looks like this

> #!/bin/sh -e
> #
> # rc.local
> #
> # This script is executed at the end of each multiuser runlevel.
> # Make sure that the script will "exit 0" on success or any other
> # value on error.
> #
> # In order to enable or disable this script just change the execution
> # bits.
> #
> # By default this script does nothing.
> iptables         -F OUTPUT
> ip6tables        -F OUTPUT
> iptables  -t nat -F OUTPUT
> iptables  -t nat -A OUTPUT -p tcp --dport 80   -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053
> iptables  -t nat -A OUTPUT -p udp --dport 8053 -d 203.0.113.1 -j DNAT --to-destination 127.0.0.1:8053
> iptables         -A OUTPUT                     -d 203.0.113.1/32  -j REJECT
> ip6tables        -A OUTPUT                     -d 2001:db8::1/128 -j REJECT
> 
> exit 0

Hypfer, 2020-01-15:
Since there a a _lot_ of working installations, there must be something wrong with your setup.

closed 2020-01-15 by Hypfer