Hypfer/Valetudo #365: Problem with persistent map data. Temporarily closed doors affecting the routing permanently.

kbehren, 2019-11-04:
FW 1886
Valetudo 0.4.0
Roborock S50

After an initial successful setup and first cleaning with all doors open the map was recorded correctly. But after the second and third cleanup with some doors closed the routing now treats the now again open doors as walls and can't reach some zones. Similar things happen with temporary obstructions like moving chairs.

One possible solution could be an option to mark doors as such or a possibility to save one master map state that is reset after each cleanup.

A map editor where obstructions can be deleted in such a master map could also be useful.

[image]

Krazy998, 2019-11-05:
On the next run it should remark those doors as passable. It does on mine. I close doors and open them all the time to control where the robot goes. Has no issues going back to areas which have a door closed.

ottelo9, 2019-11-05:
I've the same problems. My roborock Gen2 dock is under a cupboard. The robot detects the edge of the cupboard as obstacle (a line in the map)  if he moves away from the dock. Simple cleaning works, but zone cleaup and go to position doesnt work for me.

kbehren, 2019-11-05:
@ottelo9 you're right this problem occured during zone cleanup for me too. I updated the bug accordingly.

@Krazy998 I don't intend to run the complete cleanup often so that is not a real solution for me. I want to detect presence in the rooms through my home automation and clean only those that are empty.

alexjung, 2019-11-06:
I've read an article ([here](https://www.heise.de/select/ct/2019/21/1570456962754361) or in c't 21/2019, S. 174, unfortunately not free to read) where they mention the problem you describe. They say they solved it by adding the "door area" to the zone you want to clean (so the zone consists of a small rectangle drawn over the door area(s) and the actual room/area you want to clean). But it's important that you add the doors in the order of occurence starting from the docking station and ending with the actual area to clean, as the coordinates are passed in that order to the robot. 
They state that like this, the open doors are re-detected and the robot reaches the actual zone.
But I did not try this myself, just thought that it might help you.

ottelo9, 2019-11-07:
Mhh I dont know why, but sometimes it works ...
[image]

ottelo9, 2019-11-12:
> I've read an article ([here](https://www.heise.de/select/ct/2019/21/1570456962754361) or in c't 21/2019, S. 174, unfortunately not free to read) where they mention the problem you describe. They say [...]

I tried the instructions from the C'T magazin. First add door zones to my rooms, beginning from the robots dock station.
[image]

adfx, 2019-12-19:
Maybe this is helpful for you guys: [https://github.com/Thyraz/MapLoader](url)
The Maploader is triggered via mqtt and can store as much maps as you could imagine names for them.

A litle of topic:
I stuffed my S50 with that maploader because it forgot several times the complete map and with this all configured zones were not in place any longer. BUT I had to modify the maploader.js. So far it's not copying all necessary files. I'm very new to github and have no clue how to upload my version or inform the author @Thyraz about the needed change. Hints of you are very welcome (de,en)
Basically the maploader stores the map but not the orientation of the docking station which could lead into a rotated map after loading your stored map.

Belgadon, 2019-12-20:
@ottelo9 Have you found a work around for "go to position" too?
My roborock dock is under the sofa and I have the same problem.

ottelo9, 2019-12-21:
No, no fix for the go to postion.

Hypfer, 2020-01-15:
Thats beyond the scope of valetudo, sadly.

closed 2020-01-15 by Hypfer