Hypfer/Valetudo #367: automatic factory reset

Wolf6660, 2019-11-08:
Hi,
My two Xiaomi vacuum cleaners have returned to factory condition. I had to re-train him and flash it again with the Valetudo firmware.
Why did this happen?

thanks
Daniel

rickardrockard, 2019-11-08:
See #206

megusta-01, 2019-11-23:
same problem, reflashed two times. yesterday again factory settings.

stormtroopa, 2019-11-24:
I got my robot yesterday and flashed it right away with v11_001886. Everything worked fine, also cleaning. This morning the roborock started talking out of nothing: "...root failure...restoring the initial version...". 
After restoring the initial version I flashed Valetudo again. Only 10 min. later the s55 restores its initial software again.
What can I do?

rickardrockard, 2019-11-24:
Keep the discussion in #206

closed 2019-12-18 by Hypfer