Hypfer/Valetudo #368: [Feature Request] Dark Theme

silviob72, 2019-11-09:
Dark Theme for WebUI. Also with time for turn on/off automatically and with sunrise sunset, or via mqtt command.

pax2011, 2019-11-15:
It would be nice!!

rimrul, 2019-11-15:
I've done rough quick dark theme toggle with browser developer tools and the dark css that already ship with valetudo:

[image]
[image]

There's still some improvements to be made...

silviob72, 2019-11-15:
Cool! 👍

pax2011, 2019-11-15:
On chrome I use [https://chrome.google.com/webstore/detail/dark-reader/eimadpbcbfnmbkopoojfekhnkhdbieeh](url) but only on pc
[image]

notz, 2019-11-17:
It would be cool the to load own css theme per browser. So we can override the some styles if you embed valetudo in home assistant as iframe.

erkexzcx, 2019-11-17:
Someone was doing God's work by bringing dark theme.

rimrul, 2019-11-21:
New and improved state of #374 :

[image]
[image]

Hypfer, 2020-01-15:
PR is already merged to master. Thanks @rimrul

closed 2020-01-15 by Hypfer