Hypfer/Valetudo #370: DHCP bug

marv42dp, 2019-11-10:
## Describe the bug

The original firmware and valetudo seem to have a bug that prevents obtaining an IP if the DHCP server is different from the wifi ap.
My setup: 

Router: OPNSense = DHCP server, 192.168.142.x
WIFI AP1: Ubiquiti AP-AC-Mesh, SSID: magrathea
WIFI AP2 (for testing): AVM 7490, Mode: IPClient, SSID: robo

All WIFI devices do get IPs from the OPNSense DHCP, except my Xiaomi Mi Rockrobo (1st gen).
If I put the 7490 back to router mode with DHCP enabled, the robot will obtain an IP.
I did cross-check against the IP-range being the problem, by setting the 7490 to the range used by the OPNSense (no connection between the two for this test, of course).

I even dug up an old Draytek Vigor 2110n, which has an explicit DHCP-relay option, for confirmation, and it's the exact same here: as soon as the internal DHCP is off and the relay active, the robo can't obtain an IP. Other WIFI devices can.

## To Reproduce

### Screenshots

### Vacuum Model

Xiamo Mi 1st gen (rockrobo.vacuum.v1), 3468

### Valetudo Version

0.4.0

### User-Agent

Chrome

## Expected behavior

The robo should be able to obtain an IP from any DHCP server, not just from the interface it's connected to. Or we should be able to set a static IP as a workaround.

## Additional context

I've only recently replaced the 7490 with the OPNSense for WAN routing.
However, the problem did also exist in the old setup: even with the 7490 acting as DHCP server I could not connect my robo via the Unifi AP (while other devices could connect perfectly fine).

My workaround was to use the 7490s builtin WIFI for the robo, which worked just fine until I installed the OPNSense and "degraded" the 7490 to IP Client mode = no more DHCP server on the 7490.

Hypfer, 2019-11-11:
```
root@rockrobo:~# dhclient --version
isc-dhclient-4.2.4

root@rockrobo:~# wpa_supplicant -version
wpa_supplicant v2.1
Copyright (c) 2003-2014, Jouni Malinen <j@w1.fi> and contributors

root@rockrobo:~# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.3 LTS"
```

Since the robot is just using standard ubuntu 14.04 packages for everything networking-related, you might want to direct your research elsewhere.

marv42dp, 2019-11-11:
Thanks for your input. 

I just now confirmed that the dhclient-4.2.4 on Ubuntu LTS 14.04.x does indeed have a problem with obtaining an IP, if the DHCP server does not run on the AP that hosts the WIFI.

Manually assigning a fixed IP is a functional workaround. Now I'll have to figure out how to to that on the robo. 

Or could we upgrade the dhclient on the robo? Because I also cross-checked: isc-dhclient-4.4.1 on Ubuntu 19.10 does not show the same behavior, DHCP works just fine with my setup.

marv42dp, 2019-11-14:
I've narrowed the bug down to a problem between dhclient 4.2.4 and the AVM box.
The problem only occurs, if the WIFI APs LAN uplink is connected to the AVM box. As soon as I move it to a switch that's connected to my OPNSense box, all clients get IPs, even if they run dhclient 4.2.4.

closed 2019-11-14 by marv42dp