Hypfer/Valetudo #372: Zoned_cleanup limit via MQTT

darkkatarsis, 2019-11-10:
## Describe the bug
When publish payload for example:
`{ "command": "zoned_cleanup", "zone_ids": ["Bathroom", "Living room"] }` 
Everything works as it should. Two zones are enable for cleaning

This one work also:
`{ "command": "zoned_cleanup", "zone_ids": ["Bathroom", "Living room", "Bedroom"] }`

This not:
`{ "command": "zoned_cleanup", "zone_ids": ["Bathroom", "Living room", "Bedroom", "Library"] }`

This case works again:
`{ "command": "zoned_cleanup", "zone_ids": ["Bathroom", "Living room", "Library"] }`

## To Reproduce

### Screenshots

### Vacuum Model

Roborock S5

### Valetudo Version

 0.4.0

### User-Agent

Version 78.0.3904.97 (Official Build) (64-bit)

## Expected behavior

All multiple cases of zones should work
## Additional context

Is there any limitation for multi zone payloads?

rimrul, 2019-11-11:
Zones consist of rectangular areas. Zoned cleanup is limited to 5 of those areas since d8c38176 (part of version 0.3.1)

darkkatarsis, 2019-11-11:
In my example which doesn't work I used less than 5 zones. The zones are rather correct because each individual works very well, in some variations of many zones as well

My zones list
```
"Living room"
['21070', '24948', '26688', '29561']
"Bedroom"
['23879', '29597', '27833', '33029'], ['24396', '33029', '27057', '33459']
"Office"
['23768', '33530', '28129', '35747'], ['26355', '33137', '28092', '33530']
"Bathroom" 
['22142', '33923', '23805', '35747']
"Library"
['21218', '30562', '22770', '32171']
"Hallway"
['22955', '32279', '24507', '33888'], ['21624', '32600', '22918', '34102'], ['22622', '29597', '24064', '32279'], ['21292', '29561', '22622', '30169']
```

darkkatarsis, 2019-11-11:
Unless, if a given zone consists of small zones, they also count towards the limit of 5?

rimrul, 2019-11-11:
Yes, that's what I meant. Up to 5 rectangles. I.e. your hallway is already close to the limit.

darkkatarsis, 2019-11-11:
Ok, now everything is clear. Thank you

closed 2019-11-11 by darkkatarsis

rimrul, 2019-11-11:
@Hypfer how plausible is it to do the following in startCleaningZoneByCoords instead of erroring out on larger arrays:

* split the array into chunks of 5
* start cleanup for the first 5 areas
* wait for those to finish cleaning
* rinse and repeat for the next 5

Hypfer, 2019-11-11:
Getting `wait for those to finish cleaning` to work reliably will be hard I think.