Hypfer/Valetudo #378: [Enhancement] - Change Timezone UI

MrReSc, 2019-11-20:
## Enhancement
The time zone can be set under `Settings -> Timezone` via the gear icon. Unfortunately this is not very intuitive because the icon is very small. It would be better to have a gear icon under `Settings -> Info` next to the time zone.

### Screenshots
[image]

### Valetudo Version
0.4.0

pejotigrek, 2019-12-10:
just a quick question - you wrote:

> The time zone can be set under Settings -> Timezone via the gear icon. Unfortunately this is not very intuitive because the icon is very small

how can I set this timezone? I am on valetudo 0.4.0, and can't se an option [is it really THAT SMALL? ;)]. I can go to Settings->Info to see the timezone, but it can't be set

MrReSc, 2019-12-10:
The time zone can be set under `Settings -> Timers` in the top right corner.
Sorry i mean `Timers` not `Timezone` as in OP.

pejotigrek, 2019-12-10:
@MrReSc thanks for clarification! :)

Hypfer, 2020-01-15:
Sounds reasonable. I'll await your PR

closed 2021-01-09 by Hypfer