Hypfer/Valetudo #379: [Enhancement] - Set ntp Server via GUI

MrReSc, 2019-11-20:
## Enhancement
I used the image from dustbuilder. Since you can't specify a local ntp server on the website, I added my local ntp server to the file `/opt/rockrobo/watchdog/ntpserver.conf` and restarted the robot.  This works perfectly. It would be great if you could add an ntp server via GUI.

HHenning0815, 2020-01-05:
That would be very nice - because my robo has not correct time and I want to Change the NTP Server and have also a button for "sync now" would be fine in this context

pidator, 2020-01-13:
> That would be very nice - because my robo has not correct time and I want to Change the NTP Server and have also a button for "sync now" would be fine in this context

My robot had a wrong timesetting even with the correct NTP server set. 
You could check the current timezone setting by typing
```
root@rockrobo:~# date
Mon Jan 13 22:28:22 CET 2020
```

If your robot has the wrong timezone, you can set the right one with the command
`root@rockrobo:~# dpkg-reconfigure tzdata`

Hypfer, 2020-01-15:
Sounds reasonable. I'll await your PR

HHenning0815, 2020-01-23:
Thanks a lot - that works!

Titanius, 2020-03-04:
Hello, I have the problem that my time is not right.  I use Valetudo and as a time server as in installation 192.168.178.1. (my Fritzbox) The time server is active in the Fritzbox.  Unfortunately I have the time in the Roborock -1 hour when I query 
$ date
I have also used 
$ dpkg-reconfigure tzdata
to set Europe / Berlin

 Current default time zone: 'Europe / Berlin'
 Local time is now: Wed Mar 4 10:52:20 CET 2020.
 Universal Time is now: Wed Mar 4 09:52:20 UTC 2020. 

I tried this too

echo 'Europe/Berlin' > /mnt/data/rockrobo/timezone
ln -sf /usr/share/zoneinfo/Europe/Berlin /mnt/data/rockrobo/localtime

And in this File

/opt/rockrobo/watchdog/ntpserver.conf

192.168.178.1

why is the time wrong on my Roborock S50?

even a reboot did not help

rumpeltux, 2020-03-07:
So both UTC and local time are off by an hour each?
If you’re getting a time, but the time is off, chances are that your timeserver is serving a localized time instead. Have you tried using a public timeserver?

Titanius, 2020-03-10:
So both UTC and local time are off by an hour each? **YES**

If you’re getting a time, but the time is off, chances are that your timeserver is serving a localized time instead. Have you tried using a public timeserver? **No** I have blocked Roborock's web access in Fritzbox but NTP Server 192.168.178.1 works.

The Fritzbox use the NTP-Timeserver    0.europe.pool.ntp.org

And I found this on the AVM Support Page:

NTP time servers measure Coordinated Universal Time (UTC). The FRITZ! Box adds one or two hours to UTC and uses the time in Germany, Austria, Switzerland and other Central European countries (CET or CEST).

rumpeltux, 2020-03-11:
So a bug with fritzbox then. To fix it, use a public timeserver.
As a workaround: you may also set your robot timezone to UTC (may have other side-effects though).

pidator, 2020-03-11:
> So a bug with fritzbox then. To fix it, use a public timeserver.

I don't think so. Reading the statement about NTP from AVM carefully will point out, that they give first a _general_ information about NTP

> NTP time servers measure Coordinated Universal Time (UTC). 

and then they clarify specially how the  FritzBox itselves is handeling the time (like any other NTP client does):

> The FRITZ! Box adds one or two hours to UTC and uses the time in Germany, Austria, Switzerland and other Central European countries (CET or CEST).

I'm using the NTP Server of the FritzBox too without any problems, so I'm assuming the problem still in robot configuration. Perhaps running `dpkg-reconfigure tzdata` again or trying a factory reset!? Because UTC isn't an hour off, it's just the timezone, see the output of

> Current default time zone: 'Europe / Berlin'
> Local time is now: Wed Mar 4 10:52:20 CET 2020.
> Universal Time is now: Wed Mar 4 09:52:20 UTC 2020.

pidator, 2020-03-11:
@Titanius can you provide the output of the following commands?

```
root@rockrobo:~# hwclock --show
Wed Mar 11 10:28:37 2020  -1.935641 seconds
root@rockrobo:~# date -u
Wed Mar 11 09:29:43 UTC 2020
root@rockrobo:~# date
Wed Mar 11 10:30:02 CET 2020
```

Titanius, 2020-03-11:
login: Wed Mar 11 11:16:30

root@rockrobo:~# dpkg-reconfigure tzdata

Current default time zone: 'Europe/Berlin'
Local time is now:      Wed Mar 11 09:16:44 UTC 2020.
Universal Time is now:  Wed Mar 11 09:16:44 UTC 2020.

root@rockrobo:~# dpkg-reconfigure tzdata

Current default time zone: 'Etc/UTC'
Local time is now:      Wed Mar 11 09:18:11 UTC 2020.
Universal Time is now:  Wed Mar 11 09:18:11 UTC 2020.

root@rockrobo:~# hwclock --show
Tue Jan  6 00:43:51 1970  -1.085598 seconds
root@rockrobo:~# date -u
Wed Mar 11 09:19:07 UTC 2020
root@rockrobo:~# date
Wed Mar 11 09:19:13 UTC 2020
root@rockrobo:~# ntpdate -d 192.168.178.1
11 Mar 09:19:22 ntpdate[23201]: ntpdate 4.2.6p5@1.2349 Tue Jun 21 08:19:00 UTC 2016 (6)
Looking for host 192.168.178.1 and service ntp
host found : fritz.box
transmit(192.168.178.1)
receive(192.168.178.1)
transmit(192.168.178.1)
receive(192.168.178.1)
transmit(192.168.178.1)
receive(192.168.178.1)
transmit(192.168.178.1)
receive(192.168.178.1)
server 192.168.178.1, port 123
stratum 3, precision -20, leap 00, trust 000
refid [192.168.178.1], delay 0.02866, dispersion 0.00047
transmitted 4, in filter 4
reference time:    e2133957.28606b7a  Wed, Mar 11 2020 10:17:59.157
originate timestamp: e21339b5.02a2eb95  Wed, Mar 11 2020 10:19:33.010
transmit timestamp:  e2132ba1.0e55b5d8  Wed, Mar 11 2020  9:19:29.055
filter delay:  0.02963  0.02866  0.03224  0.02908 
         0.00000  0.00000  0.00000  0.00000 
filter offset: 3603.952 3603.952 3603.951 3603.952
         0.000000 0.000000 0.000000 0.000000
delay 0.02866, dispersion 0.00047
offset 3603.952970

11 Mar 09:19:29 ntpdate[23201]: step time server 192.168.178.1 offset 3603.952970 sec

root@rockrobo:~# dpkg-reconfigure tzdata

Current default time zone: 'Etc/GMT-2'
Local time is now:      Wed Mar 11 11:24:21 GMT-2 2020.
Universal Time is now:  Wed Mar 11 09:24:21 UTC 2020.

root@rockrobo:~# ntpdate -d 192.168.178.1
11 Mar 11:24:34 ntpdate[13768]: ntpdate 4.2.6p5@1.2349 Tue Jun 21 08:19:00 UTC 2016 (6)
Looking for host 192.168.178.1 and service ntp
host found : fritz.box
transmit(192.168.178.1)
receive(192.168.178.1)
transmit(192.168.178.1)
receive(192.168.178.1)
transmit(192.168.178.1)
receive(192.168.178.1)
transmit(192.168.178.1)
receive(192.168.178.1)
server 192.168.178.1, port 123
stratum 3, precision -20, leap 00, trust 000
refid [192.168.178.1], delay 0.02943, dispersion 0.00072
transmitted 4, in filter 4
reference time:    e2133957.28606b7a  Wed, Mar 11 2020 12:17:59.157
originate timestamp: e2133aec.a6cffca3  Wed, Mar 11 2020 12:24:44.651
transmit timestamp:  e2132cd8.b1d79e32  Wed, Mar 11 2020 11:24:40.694
filter delay:  0.03249  0.02943  0.03299  0.03030 
         0.00000  0.00000  0.00000  0.00000 
filter offset: 3603.953 3603.954 3603.953 3603.954
         0.000000 0.000000 0.000000 0.000000
delay 0.02943, dispersion 0.00072
offset 3603.954812

11 Mar 11:24:40 ntpdate[13768]: step time server 192.168.178.1 offset 3603.954812 sec
root@rockrobo:~#

pidator, 2020-03-11:
> Current default time zone: 'Europe/Berlin'
> Local time is now: Wed Mar 11 09:16:44 UTC 2020.
> Universal Time is now: Wed Mar 11 09:16:44 UTC 2020.

Local time is not set to CET, it's UTC.
And UTC at 11:16 CET should be 10:16, not 09:16

do first
`root@rockrobo:~# dpkg-reconfigure tzdata (change to UTC)`
again to change the local timesetting to UTC.

> root@rockrobo:~# hwclock --show
> Tue Jan 6 02:54:31 1970 -1.241569 seconds

the too old hwclock reference could be the issue for not getting the right time and the great delay by ntp. correct first the hwclock to the current local time by typing
`root@rockrobo:~# hwclock -w`
check again with
`hwclock --show`

then try again to reconfigure tzdata !

Titanius, 2020-03-22:
Thanks to all.  A fresh installation helped.  The time in Valetudo is correct now.  Query "date" and „date -u“ is -1 which is not so bad.

Hypfer, 2021-02-02:
This is _sorta_ solved with 8b54c7d0066d7b4b306aaf1b096ce1027651998c

We're the NTP Client now. This still needs UI support of course.

Handling timezone setting would be another issue to tackle 🤔

closed 2021-02-02 by Hypfer