Hypfer/Valetudo #380: [Idea, not related to the code] Stickers

white-gecko, 2019-11-25:
I would like to have a sticker which I can put on the robot to show that it runs valetudo.
I think this is important to give credit to the community, to educate visitors who see the robot, and because I also put stickers on my laptops, why not on my floor-top-computer as well.
Regarding the educational aspect: as a person who "does semething with computers" people see the devices which I buy and the software which I'm using and sometimes also ask me what do buy. So I see my self in the position to help out with media competency and data competency.

If they see the robot as is they might think it is a cool thing to buy a robotic vacuum cleaner, while they don't see the implications to there privacy, which data is collected, and where the data is going, in short under which circumstances it is a cool idea.

I don't know if some of you think about the same issues. Maybe somebody has already created a cool sticker. Maybe we can also include this privacy aspect into the sticker.

white-gecko, 2019-11-25:
[image]

closed 2019-11-26 by Hypfer

Hypfer, 2022-09-21:
Hi so first of all sorry for shutting this down back then.

I was very overwhelmed with the whole FOSS business and rather annoyed by people ignoring the "Please only open issues for actual Bugs" rule.
Fortunately, since then not only have I gotten better at handling this I think but also there are Github discussions now which allow for proper separation of Bugs and non-bugs without there being an ever increasing issue counter.

Anyway, good idea back then. Sorry again for not appreciating that.

BTT:
The current sticker situation is a follows

There are two designs that can be found here:
https://github.com/Hypfer/Valetudo/tree/master/assets/misc

The Valetudog and the "Works with Valetudo" sticker.

[image]

[image]

There was an attempt to make them available for purchase using Redbubble, however the pricing and quality both were very much subpar. I've also looked into setting up a shop and selling them myself, however that's also a lot of pain in germany so I opted to not do so.

This unfortunately means that there currently is no way of getting those apart from meeting either Dennis or Me at some conference.

_Unless_ you're located in the US that is in which case there seems to be someone selling prints of them:
https://cloudfree.shop/product/works-with-valetudo-sticker/

As the design is available as an SVG, you can also of course print a run of them on your own if you

momo42, 2023-02-11:
> [image]

> This unfortunately means that there currently is no way of getting those apart from meeting either Dennis or Me at some conference.
> 
> _Unless_ you're located in the US that is in which case there seems to be someone selling prints of them: https://cloudfree.shop/product/works-with-valetudo-sticker/

I want to add, that I successfully ordered two Stickers from Cloudfree to Germany this month. Shipping time one week or so.
Not applied yet but quality seems to be excellent. :-)

[image]