Hypfer/Valetudo #383: problems with installation on Gen1 & some questions about resetting

pejotigrek, 2019-11-27:
after a two-day battle with installation I'm taking my chance to ask the community for hints - & question about resetting the vacuum at the end of my post.

**Hardware:** Xiaomi Vacuum Gen1
**Software:** 3.5.4_00404 (according to MiHome app)

I've created my own firmware image using [Dustbuilder](https://dustbuilder.xvm.mit.edu/) and based the build on ver. 3468 (as recommended). tried to install it in three ways:

1. from my VM VirtualBox Ubuntu installation (I'm on Windows), without resetting WiFi (so using my LAN's assigned IP numbers)
2. from my Ubuntu laptop, without resetting WiFi (as above)
3. from my Ubuntu laptop, with WiFi reset, so with a direct connection laptop<->vacuum.

because I didn't build the image by myself, I skipped the _preparations_ part od the installation wiki and jumped to [Flashing the firmware image](https://github.com/Hypfer/Valetudo/wiki/Installation-Instructions#flashing-the-firmware-image).

all of the above 1-3 ways failed. I consider the 3rd option is "the most reliable/desired", so here's the console output during that try (ofcourse MY_TOKEN was the real working token):

```
user@ubuntu:~$ mirobo --ip 192.168.8.1 --token MY_TOKEN update-firmware v11_003468.pkg
Going to update from v11_003468.pkg
INFO:miio.updater:Serving on 0.0.0.0:42247, timeout 10
INFO:miio.updater:Using local v11_003468.pkg (md5: 6a10c744036959deb3446f096e17f308)
Hosting file at http://192.168.8.26:42247/v11_003468.pkg
ERROR:miio.protocol:unable to parse json '': Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "/home/user/.local/bin/mirobo", line 11, in <module>
    sys.exit(cli())
  File "/home/user/.local/lib/python3.6/site-packages/miio/click_common.py", line 57, in __call__
    return self.main(*args, **kwargs)
  File "/home/user/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/user/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/user/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/user/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/user/.local/lib/python3.6/site-packages/click/decorators.py", line 64, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/home/user/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/user/.local/lib/python3.6/site-packages/miio/vacuum_cli.py", line 609, in update_firmware
    update_res = vac.update(url, md5)
  File "/home/user/.local/lib/python3.6/site-packages/miio/device.py", line 363, in update
    return self.send("miIO.ota", payload)[0] == "ok"
  File "/home/user/.local/lib/python3.6/site-packages/miio/device.py", line 278, in send
    self.__id = m.data.value["id"]
TypeError: 'NoneType' object is not subscriptable
ERROR:miio.updater:No request was made..
user@ubuntu:~$ 
```

**tried it few times, as root user and as normal user - all with the same result. did I missed something, or I must make a factory reset - not only wifi reset?** and if so - the next question is below.

resetting WiFi (home+power button press) IS NOT resetting/clearing usage & cleaning stats. also history stays intact; **does the factory reset (involving also pin in the reset hole) clear above mentioned?** I mean - I'm a fan of the statistics and numbers, it would be sad for me to lose about two years of mileage and start from scratch ;)

Skorfulose, 2019-12-01:
Same issues for me with my Gen1. I don't know the exact SW version, but I updated it once in a while when I had it connected to the Mi app. So maybe there has been some change in recent firmwares?

hgelpke, 2019-12-02:
Same issue here
Currently on FW 3.5.00.3476

pejotigrek, 2019-12-02:
> Same issues for me with my Gen1. I don't know the exact SW version, but I updated it once in a while when I had it connected to the Mi app. So maybe there has been some change in recent firmwares?

I hope not.. I was looking forward to root & flash custom firmware :(

hgelpke, 2019-12-02:
I dug a little more into this and apparently it is an issue with 3.5. You can factory reset to the FW that came when it shipped per the wiki but I haven't done it yet. Hoping to downgrade and then apply the FW.

pejotigrek, 2019-12-02:
> I dug a little more into this and apparently it is an issue with 3.5. You can factory reset to the FW that came when it shipped per the wiki but I haven't done it yet. Hoping to downgrade and then apply the FW.

as I asked in my original post - does factory reset clean the useage/cleaning stats? I mean: brush, sensor, filter status & total time & clearing area?

Skorfulose, 2019-12-07:
I factory reset mine today. It was indeed on 3.5.0 something. Flashing the dustbuilder image with 3.3.9 3468 worked flawlessly now. Thanks for the hint hgelpke!
Total cleaned area, time and count are kept. Only the cleaning history is gone, but that's okay for me.
Go ahead guys! Don't forget to close this issue ;-)

pejotigrek, 2019-12-07:
thanks for additional info & answers!

factore reset worked like a charm. stats were kept, valetudo installed. another happy user! :)

closed 2019-12-07 by pejotigrek