Hypfer/Valetudo #384: Localizations for client's WebUI

HackintoshHD, 2019-11-30:
I'd like to propose adding localizations to the client's WebUI so that the user interface can appear in different languages based on your device's locale. I see this has already been implemented in Valetudo [here](https://github.com/rand256/valetudo/commit/63d3039da11e7d6719b0a81ce5c341667d0a583d).

rimrul, 2019-11-30:
The Valetudo RE commit could probably be adapted to Valetudo. I would import the packages([i18next](https://github.com/i18next/i18next),[i18next xhr backend](https://github.com/i18next/i18next-xhr-backend),[loc-i18next](https://github.com/mthh/loc-i18next)) via `package.json` or `yarn.lock`. There have also been some changes to Valetudo (and Valetudo RE) since ee509ca, which seems to be the latest Valetudo commit in Valetudo RE.

HHenning0815, 2020-01-05:
I would like to have localisation also (WAF and so) - please let know when and how ist possible.

Hypfer, 2020-05-15:
Duplicate of #70

closed 2020-05-15 by Hypfer