Hypfer/Valetudo #385: Problem with map always resetting

Snuffsis, 2019-12-01:
Hardware: Roborock S50
Software: Firmware version 3.3.9 
                Valetudo 0.4.0

My problem is that whenever I start a new cleanup, the map always resets, Doing Zone Cleanup seems to keep the map, but when I use "Start" for a full house clean up it will delete the map and redraw it.

I have tried with Persistant Data checked and unchecked and the problem still exists.

Is this expected behavior, or is this some setting that controls this?

roflcoopter, 2019-12-01:
It is stock functionality, nothing Valetudo can do about it.

Snuffsis, 2019-12-01:
> It is stock functionality, nothing Valetudo can do about it.

I see, thanks, then i can stop tearing my hair out of my skulle trying to fix this.
I guess my solution will be to just make one zingle zone for the entire house then.

closed 2019-12-01 by Snuffsis