Hypfer/Valetudo #386: "no map data", /etc/hosts and /etc/rc.local are reset

juantxorena, 2019-12-04:
Hardware: Roborock S55
Software: Firmware version 3.3.9, firmware build 001886
Valetudo 0.4.0

I have basically the same problem as #353 with a Xiaomi Mi Gen 2, and the solution is the same, that is to add the missing lines to /etc/hosts and /etc/rc.local.

However, I've noticed that somehow the aforementioned files are reset to their original content at some point in time, so I have to add the missing lines again. I've tried to install auditd to check who is modifying these files and when, but the kernel doesn't support it.

I will try to do a full clean again, maybe that's the reason why they are reverted. Help would be appreciated, I will provide any information that is required.

ecm75, 2019-12-12:
Have you tried setting the immutable attribute on those files?
This should prevent ANY user (even root) from changing or deleting those files.
```root@rockrobo:~# touch test
root@rockrobo:~# chattr +i test
root@rockrobo:~# rm test 
rm: cannot remove 'test': Operation not permitted
```
To regain access to the file simply remove the attribute again:
`chattr -i test`

juantxorena, 2019-12-13:
I just did that, and I suppose it'll work, but I wonder why did that happened, and how can it be avoided in the future. Any clue on where to start looking?

closed 2020-01-15 by Hypfer

matteofranceschini, 2020-03-29:
I've had the same problem (and still have it) everytime I add cleaning zones.
To be sure that zones were the issue I've started from a clean Valetudo and never added a cleaning zone. It's a month that I'm trying so and everything was good so far.

This morning I've added 3 zones from the web interface while the robot was in the dock and after that I've started a zone clean. In a matter of minutes the map was cleared showing "no map data".

As a matter of fact, my rc.local and hosts files are still Valetudo's, thanks to @ecm75 suggestion to `chattr` them. Still, the robot is not showing the map.

I guess that I'll just stop using the zones, because having them without map is not useful anyway hehe...

If you have ideas or tests to suggest, I'm open to everything.