Hypfer/Valetudo #387: Switch WiFi off by pressing a button

MalteFlender, 2019-12-13:
Is it possible to add a function to the software that reassign one of the physical buttons on the vacuum cleaner to switch its WiFi on/off? After I programmed the robot and it works fine on a day-to-day basis there is no need for it to constantly be in the WiFi anymore, unless I want to reprogram it. Having a way to switch WiFi on and off would be really great. It would save some battery life and lessens the overall electromagnetic radiation.

rimrul, 2019-12-13:
That seems quite a bit outside the scope of Valetudo.

MalteFlender, 2019-12-14:
What do you think would be a better place to implement such a feature?

rimrul, 2019-12-14:
I'd say log in via ssh, create a script like this one https://askubuntu.com/a/240516, but without the `notify-send` part (maybe play some sound there that informs you wether wifi is on or off) and figure out how to call that on button press instead of say starting zoned cleanup.

closed 2020-01-15 by Hypfer