Hypfer/Valetudo #391: S5 Max compatibility

fitim7, 2019-12-16:
Are you working on Valetudo for compatibility with S5 Max (newest vacuum cleaner)?

WahlFisch91, 2019-12-18:
That would be great!

Hypfer, 2019-12-19:
No. Why would I? I don't own an S5 Max

fitim7, 2019-12-19:
> No. Why would I? I don't own an S5 Max

Wow haha.. maybe because there are a lot of people outside there (including me) that love your project, but own a S5 Max?

ualex73, 2019-12-19:
The big question is this S5 Max is rootable, if isn't possible - Valetudo can never work. The rooting isn't done within this project.

Hypfer, 2019-12-19:
I'd happily accept pull requests.

You see, I've built valetudo for myself and then shared it, because I thought that it might be useful for other people. I didn't build it _for_ other people. I just choose an open approach so everyone can profit from it or enhance it.

Why should I spend my free time on actively researching and building something that I neither own nor use?

I'd understand these expectations if valetudo was closed-source freeware, since then there is simply no other way. But it's not. It's FOSS on purpose.

Also what @ualex73 said

fitim7, 2019-12-19:
Okay. Didnt know these things.. and since im not good in coding, maybe one day someone will work on this. Thanks anyway

ualex73, 2019-12-19:
@fitim7 the Gen1/2 rooting is described in https://github.com/dgiese/dustcloud.git ... I don't know if they will/can support the S5 Max. Recently they seem to have blocked rooting for the Gen1/2 too - with newer firmware (making the vacuum basically useless for me - if would have that firmware)

tiagofreire-pt, 2019-12-24:
> 
> 
> @fitim7 the Gen1/2 rooting is described in https://github.com/dgiese/dustcloud.git ... I don't know if they will/can support the S5 Max. Recently Xiaomi seem to have blocked rooting for the Gen1/2 too - with newer firmware (making the vacuum basically useless for me - if would have that firmware)

Here you can find rooted 1910 Gen2 firmwares, ready for download: https://github.com/zvldz/vacuum

fitim7, 2019-12-25:
> Here you can find rooted 1910 Gen2 firmwares, ready for download: https://github.com/zvldz/vacuum

How do I even know wich firmware is for my robot? Where do I get the information wich gen is mine?

tiagofreire-pt, 2019-12-25:
> How do I even know wich firmware is for my robot?

Check it with MiHome app, if it isn't not yet rooted or valetudo softmodded.

> Where do I get the information wich gen is mine?

Xiaomi 1S = Gen1
Roborock S50/S55 = Gen2
S5 Max + T6 + S6 = Gen3

jonathan-reisdorf, 2020-01-02:
My gut feeling is that S5 Max is already delivered with a firmware blocking rooting, so that even factory resetting wouldn't work for making it rootable (unless sb cracks whatever latest changes have been done to prevent rooting).
Also, it's (probably?) a Gen3 robot despite the naming, so https://github.com/dgiese/dustcloud/issues/252 applies

Hypfer, 2020-01-15:
Guess we won't see that anytime soon

closed 2020-01-15 by Hypfer

dmwyatt, 2020-12-05:
> Guess we won't see that anytime soon

Looks like it's getting closer...

https://github.com/dgiese/dustcloud/issues/213