Hypfer/Valetudo #392: Installing stock firmware (gen 1) does not work with China server anymore?

jockebq, 2019-12-20:
I tried Valetudo, but I went back to stock firmware by resetting the device. This made the device install firmware 3062.

With this firmware it works as it should (I have Xiaomi Home set to Mainland China). But when I update to 3452 and 4004 I get a notification saying “Please select local server”. This never happened before? And I am not able to reach the vacuum’s settings, neither does the map work on 3452 or 4004.

Why is this? It is not a European model I am using, it is a China version, used on the Chinese mainland server.

I am located in Europe.

Version 4004 worked before trying Valetudo, and I used Mainland china server. I don’t want to switch server as I got other chinese Xiaomi devices which requires the Chinese server.
Is it something Valetudo has broken?

### Vacuum Model

Gen 1

## Additional context

ualex73, 2019-12-20:
Check the Valetudo page:
PSA: DO NOT UPDATE YOUR GEN2 TO 1910 & GEN1 TO 4004

And if you fully factory reset it, it isn't something to do with Valetudo ...

jockebq, 2019-12-20:
@ualex73 it does not explain why you should not update to 4004? And I had 4004 before trying Valetudo, and it worked great!

ualex73, 2019-12-20:
@jockebq sorry, i don't have this information.

closed 2020-01-15 by Hypfer