Hypfer/Valetudo #393: No Webinterface

piscator83, 2019-12-21:
## Describe the bug
After rockrobo resetted himself I flashed firmware like I always do but after rockrobo tells me that the Flash was successful the webinterface is unreachable.

## To Reproduce

### Screenshots

### Vacuum Model

### Valetudo Version

### User-Agent

## Expected behavior

## Additional context

xrolfex, 2019-12-21:
You may have to reboot again, or ssh in and start the Valetudo service manually the first time.

Hypfer, 2020-01-15:
Since there a a _lot_ of working installations, there must be something wrong with your setup.

closed 2020-01-15 by Hypfer

dbeltman, 2020-03-21:
If you've been here and ran into a "dead end" like me, it might have to do with the way the firmware was built. 
If you used https://builder.dontvacuum.me/ , double check that you selected the "Preinstall Valetudo" option. After this, Valetudo's interface should show up this time.

Export33, 2020-04-13:
hi, i had the same problem.
i used `https://dustbuilder.xvm.mit.edu/pkg/s5/v11_001886.fullos.pkg` build the firmware after the robot had reset itself. the flash was successful but i couldn´t connect via the webinterface nor ssh.

using a prebuild firmware without building it myself as stated by @dbeltman worked.