Hypfer/Valetudo #395: Use robot to build wifi heatmap

homefefeld, 2019-12-22:
This is to suggest an enhancement (I didn't know where to post the requirement). 
Please let me know if you would be able to work on it.
I was wondering if we could use the robot to capture/gather wifi information (RSSI, noise, RX/TX rate) while hoovering and build a map with wifi information.
We could then push this data over MQTT and build a very accurate wifi heat map.
This would be very good to have to place the access points in the right location.

[image]

rimrul, 2019-12-23:
For some discussion on this topic see rand256/valetudo#44

Hypfer, 2020-01-15:
Sounds reasonable. I'll await your PR

Hypfer, 2020-05-15:
This is out-of-scope for the project

closed 2020-05-15 by Hypfer