Hypfer/Valetudo #396: How to start a new map? Maps get mixed up.

bl4k1st, 2019-12-23:
When using my bot I have to carry it to other rooms, because stairs. How can I start a new map, I notice the maps get mixed up frequently, the next room is put ontop of the current map resulting in a mess.

rimrul, 2020-01-08:
I can see 2 scenarios where this could happen: 

1. Pausing cleanup, carying the robot and then resuming.

2. Peristent map data is enabled.

In both cases the answer is to avoid that.

bl4k1st, 2020-01-08:
So you suggest switching off persistent map data?
If so, what is persistent map data good for?

rimrul, 2020-01-08:
For exactly the thing you don't want in your use case: Making the robot remember the old map and basing navigation partially on that.

closed 2020-01-15 by Hypfer