Hypfer/Valetudo #397: In the last 3. months the S50 crashed twice.

Wolfi311, 2019-12-27:
Hello
Basically, thanks a lot for the great program.
Works except for the following problem.
I have a S50 and have flashed the firmware "3.3.9_001856".
In the last 3. months the S50 crashed twice.
Then
1. The speaks English again.
2. The web interface is no longer accessible.

The only thing that helps is flashing the firmware again.
If you know the problem.
BR
Wolfgang

thomas-goerlich, 2019-12-27:
Hi Wolfgang,
i just encountered the same problem. While on my christmas vacation i was showing my family how i was able to vacum my living room without needing to be at my place and when i came back the roborock will not connect to my wifi anymore and speaks english (instead of german).
I hope there is a work around for this problem because i would not like to flash the roborock every couple of weeks and set everything up again and again.
Regards Thomas

thomas-goerlich, 2019-12-27:
I checked the other open issues and found out that this is a known error since at least may of this year with no solution as of yet:
https://github.com/Hypfer/Valetudo/issues/206

Wolfi311, 2019-12-29:
Hello, Thomas,
Then let us hope that the bug is found.
Thanks for the info
Wolfgang

jacekpaszkowski, 2020-01-02:
@Wolfi311 @thomas-goerlich i encounter exact same problem about every 5-6 weeks :(
hope there will be fix...

Jack

closed 2020-01-15 by Hypfer

Wolfi311, 2020-01-15:
Why Please would the treed be closed?
Please explain. 

Br

roflcoopter, 2020-01-15:
As pointed out by Thomas and the label on the issue, it's a duplicate of #206.