Hypfer/Valetudo #400: No Map Data

macbeto, 2020-01-01:
## Describe the bug

## To Reproduce

### Screenshots

### Vacuum Model

### Valetudo Version

### User-Agent

## Expected behavior

## Additional context

macbeto, 2020-01-01:
Hi, 
First, happy new year, and thank you for a very useful application.
I have installed in a Roborock S50 valetudo 0.4.0 and firmware 1886. All works really fine except the maps. 
I have been reading all the responses to the same problem in this forum and others, but still didn't solve it.
In my case /etc/hosts and /etc/rc.local is the same than the one in valetudo deployment folder. I already did several reboots with no success. Also in config.json have the map_upload_host as this:
  "allowSSHKeyUpload": true,
  "map_upload_host": "http://127.0.0.1"
I am working with Linux Mint and Firefox browser, but from the mobile I work with Opera with the same result. I disabled the LM Firewall but still didn't work.
Any help will be really appreciated.

Hoiss, 2020-01-03:
Hi,
unfortunately I can't help, but can confirm this behavior.
I installed valetudo about 2 weeks ago and already had 3 times lost the map. This is really a showstopper and always takes 1,5 hours total cleanup plus the definition of new zones.
BTW, maybe that's relevant: After every map reset the new map is built at about 90 degrees turned (?). So all zones also have to be reconfigured. Very annoying and also a showstopper for valetudo at the moment.
How can I help to solve this issue? Any logs, tests, ...?

Hoiss, 2020-01-03:
Hi,
unfortunately I can't help, but can confirm this behavior.
I installed valetudo about 2 weeks ago and already had 3 times lost the map. This is really a showstopper and always takes 1,5 hours total cleanup plus the definition of new zones.
BTW, maybe that's relevant: After every map reset the new map is built at about 90 degrees turned (?). So all zones also have to be reconfigured. 

How can I help to solve this issue? Any logs, tests, ...?

ChamesBont, 2020-01-05:
Hi,
I have the same Problem with the map data. After the first clean up there was no problem an the map was shown correctly. But after the second clean up I also get the message "no map data". 
I use Firmware "v11_003468" with 1 gen. of the Roborock.

nikkolade, 2020-01-08:
Hi,
I also tried the latest master version and I'm unable to see any map data using either web page or via MQTT. Valetudo is able to send some attributes information via MQTT but for some reason it stopped sending the png_data. I've been using the latest recommended v1886 with Gen2 S50.

I initially tried the automated image builder at https://dustbuilder.xvm.mit.edu/ and got it to work for a few days, but it hanged the roborock multiple times in a way that I had to do factory reset just to reflash it again with valetudo.

macbeto, 2020-01-09:
Yes, that is exactly my scenario, S5 Gen2 with  1886 firmware with valetudo 4.0.4 if I recall properly, I followed the installation instructions but I never got the map. I don't know if installing a firmware previous version could help.

Hoiss, 2020-01-09:
Please ensure that you disabled and enabled the persistent map feature and also reset the map. Afterwards start a full cleanup, that works for me.
Not sure if a reset between one of the steps is necessary, just try.

RoStief, 2020-01-15:
Hello, i had the same problem,
i have set up the image on windows, therefore the rc.local file had the wrong encoding.
So if you are running into the problem, and have downloaded the rc.local file under windows:

install dos2unix on the robot:
sudo apt-get install dos2unix
run dos2unix /etc/rc.local
reboot

Hoiss, 2020-01-15:
I have setup the image on Linux, never touched rc.local. It's built and can be used, but loosing the map in random intervals. ( Didn't loose it for about 8 days now, wow quite happy 😉)

Hypfer, 2020-01-15:
Since there a a _lot_ of working installations, there must be something wrong with your setup.

closed 2020-01-15 by Hypfer

timsn, 2020-01-15:
I can also confirm this behaviour. It's loosing the map data in random intervals and sometimes the newly created maps are rotated by 90 degrees. I just build and flashed the firmware and did no further manual changes.

Hoiss, 2020-01-15:
@Hypfer , I really don't appreciate your reaction. I hoped you somehow act more professional and want to look for the real reason for this behavior.
Various users confirm the problem, I fully followed the installation instructions without ANY adaptions.
Tell us how we can help to border the problem instead of ignoring it. 
Otherwise you could also simply close #206 to get rid of it. The participants might just have done something wrong ...  To be honest I'm just waiting to run into the reset issue, too.

Please re-open this issue!

Hypfer, 2020-01-15:
I really don't appreciate a lot of things that happend to this projects issue tracker & community.

`Acting more professional` seems to be equivalent working for free to fix other peoples problems I don't have/care about and already tried debugging for a long time + provide even more basic tech support in my free time.

It's open source for a reason. Go fix it yourself or hire someone to do it for you

Hoiss, 2020-01-15:
Oh wow, you're really grumpy.
I am enjoying valetudo and I am also willing to investigate time to improve it. But this is not possible if you don't react on questions how to help. If you have any idea how to boarder this issue, tell me / us! I assume you're the one with the deepest knowledge, even anyone can have a look at the sources.
If you just insist on ignoring user issues and simply closing existing issues I really regret to have spent my money for this project.

Hypfer, 2020-01-15:
> I really regret to have spent my money for this project

Here are a few definitions of the word donation 

> Donation is the act by which the owner of a thing voluntarily transfers the title and possession of the same from himself to another person, without any consideration; a gift.

> donate. To donate means to give something — money, goods, or time — to some cause, such as a charity. The word has a more altruistic meaning than does simply "giving"; it suggests that you don't expect anything in return for the contribution.

I never asked for those. I just put a single link hidden in a menu that is hidden most of the time for people who wanted to say `thank you` for the current state which already took at least a few hundred hours of work and is available for free to anyone in the world with the code open as well.

If you were under the impression that this way a payment to keep the project going I'm sorry but that's not the case.

Also even if that would've been the case, 10€ would only cover 5-10 minutes. Thats about the amount of time these few replies took so I'm unsure what you were expecting, really.

Hypfer, 2020-01-15:
pluralis modestiae

https://github.com/Hypfer/Valetudo/graphs/contributors

RoStief, 2020-01-15:
maybe to come back at the topic of problem solving:
did you check if the contents of rc.local are actually used?
you can do this with the command

iptables-save

The output should look like following:

root@rockrobo:~# iptables-save
# Generated by iptables-save v1.4.21 on Wed Jan 15 18:07:02 2020
*raw
:PREROUTING ACCEPT [146139:34509973]
:OUTPUT ACCEPT [156691:78271808]
COMMIT
# Completed on Wed Jan 15 18:07:02 2020
# Generated by iptables-save v1.4.21 on Wed Jan 15 18:07:02 2020
*nat
:PREROUTING ACCEPT [3959:280612]
:INPUT ACCEPT [550:47854]
:OUTPUT ACCEPT [2395:145697]
:POSTROUTING ACCEPT [2397:145817]
-A OUTPUT -d 203.0.113.1/32 -p tcp -m tcp --dport 80 -j DNAT --to-destination 127.0.0.1:8053
-A OUTPUT -d 203.0.113.1/32 -p udp -m udp --dport 8053 -j DNAT --to-destination 127.0.0.1:8053
COMMIT
# Completed on Wed Jan 15 18:07:02 2020
# Generated by iptables-save v1.4.21 on Wed Jan 15 18:07:02 2020
*mangle
:PREROUTING ACCEPT [146139:34509973]
:INPUT ACCEPT [142726:34275903]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [156697:78272632]
:POSTROUTING ACCEPT [156852:78286977]
COMMIT
# Completed on Wed Jan 15 18:07:02 2020
# Generated by iptables-save v1.4.21 on Wed Jan 15 18:07:02 2020
*filter
:INPUT ACCEPT [142727:34275943]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [156699:78273700]
-A INPUT -p udp -m udp --dport 6665 -j DROP
-A INPUT -p tcp -m tcp --dport 6665 -j DROP
-A OUTPUT -d 203.0.113.1/32 -j REJECT --reject-with icmp-port-unreachable
COMMIT
# Completed on Wed Jan 15 18:07:02 2020

Check also the same for ipv6:

root@rockrobo:~# ip6tables-save
# Generated by ip6tables-save v1.4.21 on Wed Jan 15 18:07:48 2020
*raw
:PREROUTING ACCEPT [1445:267510]
:OUTPUT ACCEPT [16:2851]
COMMIT
# Completed on Wed Jan 15 18:07:48 2020
# Generated by ip6tables-save v1.4.21 on Wed Jan 15 18:07:48 2020
*mangle
:PREROUTING ACCEPT [1445:267510]
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [16:2851]
:POSTROUTING ACCEPT [0:0]
COMMIT
# Completed on Wed Jan 15 18:07:48 2020
# Generated by ip6tables-save v1.4.21 on Wed Jan 15 18:07:48 2020
*filter
:INPUT DROP [0:0]
:FORWARD DROP [0:0]
:OUTPUT DROP [16:2851]
-A OUTPUT -d 2001:db8::1/128 -j REJECT --reject-with icmp6-port-unreachable
COMMIT
# Completed on Wed Jan 15 18:07:48 2020

Olsenius, 2020-01-21:
I also had problems with the map data suddenly disappearing without thinkering with the robot for weeks. 

Both /etc/hosts and /etc/rc.local was ok. 
Running the command `iptables-save` and `ip6tables-save` fixed it! Thanks @RoStief

Hacki1111, 2020-01-28:
I have the same problem. I´ve done many factory resets, tested different firmwares and versions of valetudo, read every single issue and tested everything but nothing works.
The outputs of both iptable commands are okay too.
The map files in the filesystem are stored after cleanup and when I disable the entrys in the rc.local and in the hosts file and connect the mi home app the map is working in the app so I guess that the stored files should be okay.
But the webinterface only shows "no map data". I´m trying for two days now and its very frustrating.