Hypfer/Valetudo #402: how to edit config files

HHenning0815, 2020-01-05:
Hello,

and thanks for Project.
Sorry for that Question, but how can the config files be edited?
I tried pico, nano… no Editor present that I found.
After searching in wiki an net - didnt find anything, just "Edit" this and that file, not how…

Regards

stcbus, 2020-01-06:
I installed nano via the package manager:

apt update
apt install nano

Hypfer, 2020-01-07:
You have to use vim

```
root@rockrobo:~# whereis vim.tiny
vim: /usr/bin/vim.tiny /etc/vim /usr/share/vim
```

Runing apt update is not a good idea since space on this machine is very limited

Lockzi, 2020-01-07:
@Hypfer I believe you can just run` apt clean` afterwards to remove the package sources

Hypfer, 2020-01-07:
Did you try it? iirc that doesn't fully solve the space issues

Lockzi, 2020-01-07:
`root@uno:~# df -h
Filesystem           Size  Used Avail Use% Mounted on
/dev/root            494M  479M     0 100% /
devtmpfs             202M  4.0K  202M   1% /dev
none                 4.0K     0  4.0K   0% /sys/fs/cgroup
tmpfs                 30M  4.0K   30M   1% /tmp
none                  50M  240K   50M   1% /run
none                 5.0M     0  5.0M   0% /run/lock
tmpfs                100M  1.6M   99M   2% /run/shm
none                 100M     0  100M   0% /run/user
/dev/mmcblk0p1       1.6G  168M  1.3G  12% /mnt/data
/dev/mmcblk0p11       16M  2.2M   13M  15% /mnt/reserve
/dev/mmcblk0p10      494M  285M  185M  61% /mnt/updbuf
/dev/mmcblk0p6        62M   14M   46M  23% /mnt/default
//10.0.1.6/Rockrobo  3.6T  1.7T  1.9T  47% /mnt/Rockrobo
root@uno:~# apt-get clean
root@uno:~# df -h
Filesystem           Size  Used Avail Use% Mounted on
/dev/root            494M  430M   39M  92% /
devtmpfs             202M  4.0K  202M   1% /dev
none                 4.0K     0  4.0K   0% /sys/fs/cgroup
tmpfs                 30M  4.0K   30M   1% /tmp
none                  50M  240K   50M   1% /run
none                 5.0M     0  5.0M   0% /run/lock
tmpfs                100M  1.6M   99M   2% /run/shm
none                 100M     0  100M   0% /run/user
/dev/mmcblk0p1       1.6G  168M  1.3G  12% /mnt/data
/dev/mmcblk0p11       16M  2.2M   13M  15% /mnt/reserve
/dev/mmcblk0p10      494M  285M  185M  61% /mnt/updbuf
/dev/mmcblk0p6        62M   14M   46M  23% /mnt/default
//10.0.1.6/Rockrobo  3.6T  1.7T  1.9T  47% /mnt/Rockrobo
`

I believe it does? How does your df -h look? I've installed CIFS and nano on my install.

NNMavy, 2020-01-07:
As your output shows it did not solve the space issue: **dev/root 494M 479M 0 100% /** the root filesystem should never be full as that will cause major issues for the os.

Lockzi, 2020-01-07:
root@uno: ~ # apt list --installed | wc -l
WARNING: apt does not have a stable CLI interface yet. Use with caution in scripts.
W: Ignoring Provides line with DepCompareOp for package gdb-minimal
W: You may want to run apt-get update to correct these problems
**284**

I am guessing it's only due to that I have more packages installed.

The only drawback I see is that it is limiting the writable area and thus shredding the last 39 mb more than your 81 mb. However, this space is only used essentially for packages and system logs.

closed 2020-01-15 by Hypfer