Hypfer/Valetudo #404: Not getting state via MQTT (always set to "UNKNOWN STATE CODE") but web UI works fine

aaronk6, 2020-01-08:
## Describe the bug

I hope I’m not missing something obvious here, but I’m not getting the vacuum’s state via MQTT. It is always set to `UNKNOWN STATE CODE`.

The vacuum reacts to commands sent via MQTT though.

For example, this works:

```
mosquitto_pub -h homeserver -t 'valetudo/rockrobo/command' -m 'locate'
```

But the state reported via MQTT looks like this:

```
{
  "cleanTime": "38.9",
  "cleanArea": "2064.2",
  "cleanCount": 87,
  "last_run_stats": {
    "startTime": 1578506467000,
    "endTime": 1578507542000,
    "duration": 1075,
    "area": "18.1",
    "errorCode": 0,
    "errorDescription": "No error",
    "finishedFlag": true
  },
  "mainBrush": "261.0",
  "sideBrush": "161.0",
  "filter": "111.0",
  "sensor": "6.6",
  "state": "idle",
  "valetudo_state": {
    "id": -1,
    "name": "UNKNOWN STATE CODE"
  }
}
```

(I’ve pretty-printed this to avoid the big horizontal scroll bar here.)

## To Reproduce

Just start listening to Valetudo’s MQTT messages:

```
# mosquitto_sub -v -h homeserver -p 1883 -t 'valetudo/#'

valetudo/rockrobo/attributes {"cleanTime":"38.9","cleanArea":"2064.2","cleanCount":87,"last_run_stats":{"startTime":1578506467000,"endTime":1578507542000,"duration":1075,"area":"18.1","errorCode":0,"errorDescription":"No error","finishedFlag":true},"mainBrush":"261.0","sideBrush":"161.0","filter":"111.0","sensor":"6.6","state":"idle","valetudo_state":{"id":-1,"name":"UNKNOWN STATE CODE"}}
```

### Screenshots

The status is properly displayed on the web frontend:

[image]

But not published via MQTT:

[image]

### Vacuum Model

Gen2

### Valetudo Version

0.4.0

## Expected behavior

Seeing the status (e.g. “Charging”) in the MQTT output.

## Additional context

Log output:

```

# tail -f /var/log/upstart/valetudo.log
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Connected successfully to mqtt server
Got token from handshake: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Loading configuration file: /mnt/data/valetudo/config.json
Dummycloud is spoofing 203.0.113.1:8053 on 127.0.0.1:8053
Webserver running on port 80
Connected successfully to mqtt server
Got token from handshake: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Connected successfully to mqtt server
```

_(Note that I’ve obfuscated the token.)_

Valetudo config:

```
{
  "spots": [],
  "areas": [],
  "mqtt": {
    "enabled": true,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "mqtt://homeserver",
    "mapSettings": {
      "drawPath": true,
      "drawCharger": true,
      "drawRobot": true,
      "border": 2,
      "scale": 4
    }
  },
  "dummycloud": {
    "spoofedIP": "203.0.113.1",
    "bindIP": "127.0.0.1"
  },
  "httpAuth": {
    "enabled": false,
    "username": "valetudo",
    "password": "valetudo"
  },
  "allowSSHKeyUpload": true,
  "map_upload_host": "http://127.0.0.1"
}
```

roth-m, 2020-01-10:
Same here with an unprovisioned (never ever connected to Internet) S50 (using firmware 1898).

aaronk6, 2020-01-11:
@roth-m, that’s interesting. Mine is also unprovisioned and sitting in a VLAN without any Internet connectivity (it can only reach a local NTP server and the MQTT broker). I’m on firmware build 1886.

roth-m, 2020-01-11:
What is strange to me is that the state is reported properly in the Web UI but not in MQTT.
It looks like events (here "miio.status") are not propagated properly.
`MqttClient.prototype.updateAttributesTopicOnEvent` is never called in [lib/MqttClient.js](https://github.com/Hypfer/Valetudo/blob/master/lib/MqttClient.js).

roth-m, 2020-01-13:
Even if the WiFi was set at firmware level (--unprovisioned=wpa2psk --ssid=xxxx --psk=xxxx), I did set it in Valetudo (Settings -> WiFi) again and got everything working properly (even the map which was also not working).

Hypfer, 2020-01-15:
`miio_client` seems to just don't connect to the cloud at all if the robot is unprovisioned which does make sense since there would normally be no cloud to connect to at all.
Both MQTT as well as the Map view use the cloud interface and therefore don't receive data.

Since the Web UI was developed first when there was no mock cloud interface in valetudo, it uses the local interface which is always available to poll its data.

While technically not being a bug, it's still confusing and could definitely be reimplemented in a a more resilient fashion

luke-barnett, 2020-02-06:
@Hypfer I have the same situation which has occurred after a decent period of it working as expected. What's the course to resetting it all to work? Restart the unit?

Hypfer, 2020-04-12:
Is this still an issue? 🤔

Hypfer, 2020-05-01:
I guess not

closed 2020-05-01 by Hypfer