Hypfer/Valetudo #405: Disabling sending map data over MQTT not working

LucaKaufmann, 2020-01-12:
## Describe the bug

I've noticed some issues with Node-RED when Valetudo is sending the map data over MQTT. I tried removing the map parts of the config and adding "provideMapData": false, but Valetudo still seems to send the map data to MQTT in the map_data topic. What's the correct way to disable that?

Here are the config examples I've tried:

`  "mqtt": {
    "enabled": true,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "mqtt://user:pw@123.123.12.12",
    "provideMapData": false,
    "caPath": ""
  },`

and 

`  "mqtt": {
    "enabled": true,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "mqtt://user:pw@123.123.12.12",
    "caPath": ""
  },`

Any help is appreciated!

rumpeltux, 2020-02-16:
Code looks fine. It should work. Can you still repro? Are you sure your config is actually being used? (e.g. change topic prefix).

LucaKaufmann, 2020-02-16:
I switched to the Valetudo RE fork and started with a fresh config which fixed the issue, so it was most likely just some config problem. Closing this one

closed 2020-02-16 by LucaKaufmann

rvdblink, 2020-04-15:
Well, I am still on Valetudo and can confirm your issue. I have the map off in the mqtt set up and still mosquitto on my docker instance get's swamped with the map data... And that is crashing my raspberry pi 3b+ in the end.

`"mqtt": {
    "enabled": true,
    "identifier": "rockrobo",
    "topicPrefix": "valetudo",
    "autoconfPrefix": "homeassistant",
    "broker_url": "mqtt://rockrobo:valetudo@10.0.1.201",
    "provideMapData": false,
    "caPath": ""
  },
`