Hypfer/Valetudo #406: Clean Passes for a zone

arieltools, 2020-01-12:
Not sure this is a bug or more like a feature request. In the Xiaomi app the user can set the amount of passes to be done in a certain zone. I can't find this feature in the current Valetudo iteration.

## To Reproduce
Go to the web interface and there's nowhere to set that.

### Screenshots

### Vacuum Model
Gen2

### Valetudo Version
0.4.0

### User-Agent

## Expected behavior
Once starting the Zone cleanup, allow the user to be able to select the amount of passes (times) that the vacuum would clean that zone.

## Additional context

rimrul, 2020-01-15:
IIRC, you can select multiple zones per cleanup. I can see 3 ways this feature (once implemented) could work, UX wise:

* Define passes per Zone when defining the zones (what Valetudo RE seems to do)
* Input a number of passes for each zone when starting the cleanup
* Input one number for the whole cleanup when starting the cleanup

I feel like option 1 is rather inflexible, but good if you know you always want multiple passes per zone for certain zones. Option 2 is probably the most flexible, but could easily get annoying rather quickly.

`api/start_cleaning_zones_by_coords` already accepts a parameter `iterations` (can be specified per area), but `api/start_cleaning_zones_by_names` doesn't, so we'd need to pass that parameter through there for option 2 & 3.

Hypfer, 2020-05-15:
Superseded by #521

closed 2020-05-15 by Hypfer