Hypfer/Valetudo #409: Wifi problem

ricruz, 2020-01-16:
After rooting the Gen 1 can't connect to any wifi.
Any help?

homefefeld, 2020-01-17:
did you configure your wifi ? is that your first install ?

reset the wifi (normal procedure), connect with a computer to the roborock wifi, then with a browser connect to 192.168.8.1 , then cha ge the wifi in the setting.

G/

ricruz, 2020-01-17:
Already done that and can't connect

ricruz, 2020-01-17:
> did you configure your wifi ? is that your first install ?
> 
> reset the wifi (normal procedure), connect with a computer to the roborock wifi, then with a browser connect to 192.168.8.1 , then cha ge the wifi in the setting.
> 
> G/

Already done that and can't connect

MrReSc, 2020-01-19:
The robot only supports 2.4GHz Wifi networks. Maybe this is your problem.

rumpeltux, 2020-02-16:
Doesn’t seem like a bug in Valetudo. You can try to get help on roboter-forum.com.

closed 2020-02-16 by rumpeltux