Hypfer/Valetudo #410: Installation guide is sligthly borked

15goudreau, 2020-01-17:
## Describe the bug

## To Reproduce

### Vacuum Model
All 

### Valetudo Version
All

## Expected behavior

15goudreau, 2020-01-17:
Second issue I encountered when building the firmware is that it just will straight up ignore the ../valetudo directory when building the firmware. When it is installed valetudo isn't actually installed. I had to install valetudo with the roborock control center app instead.

Hypfer, 2020-01-19:
Which guide do you mean? There is no git clone of this repo in https://github.com/Hypfer/Valetudo/wiki/Installation-Instructions

15goudreau, 2020-01-19:
Hi, I understand there is no git clone command in that guide. However the instructions (for someone who doesn't know linux super well) aren't clear as to how to wget the latest valetudo binary. It just says to use the url with /latest, but if you wget that link it pulls nothing but a like 50 kb file called latest. I think an example url would be more useful than 

```

Download the latest valetudo binary from https://github.com/Hypfer/Valetudo/releases/latest

wget https://github ...
```

Hence why I used git clone to pull the lastest binary, but then the naming convention complained.

herrwusel, 2020-01-20:
`https://github.com/Hypfer/Valetudo/releases/latest/download/valetudo` should fix the issue of not downloading the actual binary.

Hypfer, 2020-01-20:
Fixed. Thanks @herrwusel

closed 2020-01-20 by Hypfer