Hypfer/Valetudo #411: Support for new firmware version?

tobsch, 2020-01-19:
Xiaomi just released a new firmware version that officially introduces room naming and identification features (002008). I am very keen to give it a try as I guess it will improve the overall experience a lot. 
Is it a lot of work to support it with Valetudo? Are there plans to do so?

Hypfer, 2020-01-20:
I did take a (~4h) look at map data from an S6 and was unable to properly parse it so there's definitely some work to be done. Next, we would also need to find out how to properly integrate that with the existing features. Is it a replacement? How can we support both old as well as new modes?

There's also much more work to be done in the image building process since it got trimmed down quite a bit as well as extended with ssh and local OTA blocking. We're at a point where it would make sense to just provide prebuild images but we can't do that because there might be legal issues and I'm not particularly fond of that.

Personally, I didn't even need the virtual walls for a long time so there's no real motivation to upgrade from a working system and put more time into it. Still, if anyone was eager to implement support I would definitely support it.

Overall, the user reports for the feature I've heard were rather a mixed bag though.

Ubimo, 2020-01-20:
I would love to see that feature, mostly because the robot can now change it's direction during cleaning depending on the room (longest wall). It would definitly be faster when cleaning different rooms on one floor.

tobsch, 2020-01-20:
One issue I have right now with valetudo: My base often is moved a few cm - does the new version somehow realize that this was the case and recognizes the same map instead of starting a new one?

Hypfer, 2020-01-20:
No idea. Thats the roborock blackbox

rumpeltux, 2020-02-16:
The viomi devices also have named rooms ((partially) implemented in [ViomiMapParser.js](https://github.com/Hypfer/Valetudo/blob/viomi/lib/ViomiMapParser.js)). Not sure if any of the map format is similar or the same for the new roborocks though.

Hypfer, 2020-05-15:
Initial support exists since 0.5.0

closed 2020-05-15 by Hypfer