Hypfer/Valetudo #416: Addition to FAQ: If installation fails, do factory reset

JPT77, 2020-01-26:
I wanted to test the (used) robot before installing Valitudo. So I installed MiHome and it made an update to the firmware. After that, installation aborted just before downloading. Solution was to do a factory reset via plug- and reset-button.

Could you please add this solution to the FAQ or troubleshooting page?

thanks

btw. while downloading I get the error "unable to discover a device at ....8.1". Progress stayed at 1%.
I moved the device from the dock to directly at the laptop.  
Later I pressed Ctrl-C to abort, but it did not abort. Suddenly the progress-bar reappeared and continued at 30+% 
Can anything be broken by this process canceled?

It's very, very slow. U-Boot, eh?

JPT77, 2020-01-28:
I tried to edit the wiki myself... and create a pull request. But I simply don't get the more complex git stuff. so here you are:

## Firmware image installation fails
### before the download bar appears:

 * Token wrong? Did you initiate a WiFi reset on the robo? Then you have to refetch the token.
 * wrong IP adresses?
 * Your PC does not know how to route, more than one network interfaces active?

### after the download bar appeared:

 * Did you make an update via the Xiaomi App? Then go back to original using factory reset: while holding the plug button shortly press the reset button

Hypfer, 2020-02-10:
I've unlocked the wiki 

They were locked for vandalism. Lets hope that that will not happen again

JPT77, 2020-02-13:
Thanks. Added this to the installation instructions as well.

closed 2020-02-13 by JPT77