Hypfer/Valetudo #419: Map and Wifi Led issues with special characters in the wifi ssid

Hacki1111, 2020-01-29:
## Describe the bug
My Wifi network has a " \  "and a " / " in the ssid but the network works fine on all my devices even on the robot with stock firmware and the original app.
But with valetudo the map isn´t working and the wifi led blinks all the time.
Nevertheless the control funktions and option are working.
The robot creates the correct map files.
With a guest wifi which has a name with only letters everything works fine but I want to use the robot in my normal network and don´t want to rename it.
It must be a problem with valetudo.

## To Reproduce

Change the ssid of the wifi ap

### Vacuum Model

S50 (gen2)

### Valetudo Version

0.4.0 but also tested on 0.3.1 

## Expected behavior

All functions working even with those characters in the ssid

rimrul, 2020-01-30:
I think the Backslash is the problematic character here. We'd probably need to run `JSON.Parse(req.body)` before passing the credentials to `vacuum.configureWifi()` in `/api/wifi_configuration`

Hypfer, 2020-02-10:
I couldn't believe it at first but this is indeed a bug.

> The 802.11 standards prior to the 2012 edition did not define any particular encoding/representation for SSIDs, which were expected to be treated and handled as an arbitrary sequence of 0–32 octets that are not limited to printable characters. The IEEE 802.11-2012 defines a tag that the SSID is UTF-8 encoded and when interpreting could contain any non-ISO basic Latin characters within it. Wireless network stacks must still be prepared to handle arbitrary values in the SSID field.

[https://en.wikipedia.org/wiki/Service_set_(802.11_network)#Service_set_ID_(SSID)](https://en.wikipedia.org/wiki/Service_set_(802.11_network)#Service_set_ID_(SSID))

Hypfer, 2020-04-12:
Honestly, this seems harder than anticipated
```
> JSON.stringify("Hallo\Welt")
'"HalloWelt"'
> JSON.stringify("Hallo\\Welt")
'"Hallo\\\\Welt"'

> JSON.parse('"Hallo\Welt"')
'HalloWelt'
> JSON.parse('"Hallo\\Welt"')
SyntaxError: Unexpected token W in JSON at position 7
> JSON.parse('"Hallo\\\Welt"')
SyntaxError: Unexpected token W in JSON at position 7
> JSON.parse('"Hallo\\\\Welt"')
'Hallo\\Welt'
```

Hypfer, 2020-05-31:
Funfact: Roborock also has some issues with "special" characters in the SSID

`get_network_info` returns

```
{
	'rssi': -45,
	'mac': '78:11:dc:ff:ff:ff',
	'ip': '192.168.x.x',
	'ssid': 'Das',
	'bssid': '04:bd:88:ff:ff:ff'
}
```

but the ssid is actually `Das IoT`

kbeilix, 2020-07-11:
Similar issue here with S50 Gen2. My SSID has a "-" character and the password contains several special characters. I was able to connect to my wifi on valetudo 0.4.0 but fail to achieve this in 0.5.3
I tried the following steps:

- connected to the AP of robot and entered the SSID and PW in the Webinterface --> AP of the robot disappears but does not connect to my wifi

- tried to use rrcc to enter the wifi data --> robot AP remains and no change is observed
- shell command mirobo.... withe PW in double quotes ("PW")
- update from 0.4.0 to the recent version with mirobo command --> update works but does not connect to my wifi afterwards
- tried to connect through webinterface on 0.5.3 to a hotspot without special characters --> works like expected

Any ideas? I also tried to enter the wifi data through SSH but was not sure how to proceed

Hypfer, 2020-07-11:
@kbeilix 

Did the underlying roborock firmware version change? Because I'm not aware of any changes in valetudo.

kbeilix, 2020-07-11:
Thanks for the quick reply. You are right that the underlying firmware version has changed. The 0.4.0 valetudo was created by dustbuilder and has firmware version 3.3.9. The one I was trying to update to was generated by dustbuilder as well yesterday (2020-07-10).
I would like to look into this a little more.  However I was not able to identify the location where the password is stored on the device. Can you give me an indication on where to start?

Hypfer, 2021-01-12:
I don't think that this will ever be solved, sorry.

As a workaround, please use a password without `\`

closed 2021-01-12 by Hypfer